Moddable-OpenSource/moddable #125: Cannot build example/piu/wifi-config for esp

gusdgg, 2019-02-11:
When I try to build the Wifi-Config example for esp8266 in windows, I got the following error message:

libxslib.a(xsNumber.o): In function `fxNewNumberInstance':
         xsNumber.c:83: undefined reference to `fxBigIntToNumber'
libxslib.a(xsNumber.o): In function `fx_Number':
         xsNumber.c:220: undefined reference to `fxBigIntToNumber'
collect2: error: ld returned 1 exit status

It's seems that the windows build is missing a required file. I've fix the issue editing the "%MODDABLE%\tools\mcconfig\nmake.esp.mk" make file and adding the required "xsBitInt.o" reference in the XS_OBJ section:

XS_OBJ = \
	$(LIB_DIR)\xsAll.o \
	$(LIB_DIR)\xsAPI.o \
	$(LIB_DIR)\xsArguments.o \
	$(LIB_DIR)\xsArray.o \
	$(LIB_DIR)\xsAtomics.o \
	**$(LIB_DIR)\xsBigInt.o \**
	$(LIB_DIR)\xsBoolean.o \

lprader, 2019-02-11:
Yep, sorry about that. It's also missing the RegExp file (see [this thread](https://github.com/Moddable-OpenSource/moddable/issues/119)). Your fix is correct.

We're planning to push an update to the repository to fix all of that soon.

phoddie, 2019-02-14:
Fix pushed.

closed 2019-02-14 by phoddie