Moddable-OpenSource/moddable #126: Cannot build examples piu/one-line, piu/one-line-keyboard: module "socket" not found

lll000111, 2019-02-12:
Examples in examples/socket work. (*Incidentally, how do I actually see anything? The demo runs but that's just a black screen. How do I see a console?*)

```
$ mcconfig -d -m -p lin
....
# xsl modules
# mcrez resources
# exception: require: module "socket" not found!
### ReferenceError: require: module "socket" not found!
/home/mha/Projects/moddable/build/tmp/lin/debug/mc/one-line-keyboard/makefile:382: recipe for target '/home/mha/Projects/moddable/build/tmp/lin/debug/mc/one-line-keyboard/mc.xs.c' failed
make: *** [/home/mha/Projects/moddable/build/tmp/lin/debug/mc/one-line-keyboard/mc.xs.c] Error 22
make: *** Waiting for unfinished jobs....
Total resource size: 28535 bytes
```

phoddie, 2019-02-12:
> Examples in examples/socket work

It appears that the build failed because the `socket` module is not available on the Linux build.

> How do I see a console

Use the debugger, as described in the  [Getting Started](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md) materials and detailed in the [xsbug documentation](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/xsbug.md).

Please do keep in mind that the Moddable SDK is focused on creating applications for embedded micro-controllers. While many of the examples do run on desktop platforms including macOS, Windows, and Linux, not all of the examples do.

closed 2019-02-12 by phoddie

lll000111, 2019-02-12:
Is there an emulator? I have boards, but especially now during evaluation and getting stuff to run at all - I have to rewrite some APIs, e.g. file system access or websockets to use the ones provided by moddable - being able to test-run on the PC is much faster than having to write to the board each time.

phoddie, 2019-02-12:
> Is there an emulator?

There is a simulator, which is how "many of the examples...run on desktop platforms" as noted above. The simulator is built as part of the Moddable tools build documented in [Getting Started](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md).

lll000111, 2019-02-12:
I'm already using that, but it still is "Linux" as target? I followed the instructions for "Linux" and it says (emphasis added)

> Verify the host environment setup by building the starter helloworld application **for the desktop simulator target:**

So pardon me, but if it's a simulator the stuff that's not available on "Linux" should work because I'm actually running on the simulated platform, no?

When I run e.g. Android development I can run Android code on "Linux" or "Windows" just fine - in the emulator/simulator. As far as I'm concerned I'm not *actually* running anything at all on "Linux" so I'm a little confused about what is what here, since the "Linux" instructions indicate that I'm running code in the simulator. Which should not have the "Linux" platform limitations.

lll000111, 2019-02-12:
Furthermore

> It appears that the build failed because the `socket` module is not available on the Linux build.

As I wrote, I can run other "socket" importing examples just fine, e.g. examples/network/socket/socketlistener

phoddie, 2019-02-12:
> So pardon me, but if it's a simulator the stuff that's not available on "Linux" should work because I'm actually running on the simulated platform, no?

That is not correct. The simulator uses the host platform, in this case Linux, to provide functionality such as a file system and network. As there are not implementations of the `File` and `Socket` modules for Linux, they are unsupported today on the simulator hosted on Linux.

> As I wrote, I can run other "socket" importing examples just fine, e.g. examples/network/socket/socketlistener

The [manifest_net.json](https://github.com/Moddable-OpenSource/moddable/blob/public/examples/manifest_net.json) file is where the core network related modules are built, include `Socket`.  There is no `platform` section  for the `lin` target there. Consequently, it is unclear how you confirmed the examples using `socket` are running on the Linux target.

lll000111, 2019-02-12:
> Consequently, it is unclear how you confirmed the examples using `socket` are running on the Linux target.

The simulator started and there was no error.

lll000111, 2019-02-13:
I see, the exception is during RUNTIME for the socket examples but not during compile time.

So for some examples I get a "no socket module" error during compile time, for others only when I use xsbug to see exceptions occurring during runtime. Hmmm.

phoddie, 2019-02-13:
In JavaScript, there are different kinds of errors. Some are classified as "early errors" which means that they are flagged when the JavaScript is compiled, while others are deferred to runtime.

lll000111, 2019-02-13:
Yes I know, I just find it funny that this is the same issue &mdash; AND it looks to be compile time to me in every case. After all, there's the require statement to load the socket module, certainly something that I would have expected to be picked up during compile time of the project.

phoddie, 2019-02-13:
> ...there's the require statement to load the socket module

Would you mind sharing the specific instance of the "`require` statement" you are referring in this instance?

> ...certainly something that I would have expected to be picked up during compile time of the project.

Note that `require` is not a statement. It is a function call. Consequently, its arguments may be variables determined at runtime. Function calls are not evaluated at compile time unless they are contained in, or called by, the body of a module which is preloaded.

lll000111, 2019-02-13:
_By the way, this is all an aside at this point, for me, but interesting to know._

This is where I saw the error only after running it in the debugger:

https://github.com/Moddable-OpenSource/moddable/blob/public/examples/network/socket/socketlistener/main.js#L15

This is from the original comment above:

https://github.com/Moddable-OpenSource/moddable/blob/public/examples/piu/one-line-keyboard/main.js#L18

which includes

https://github.com/Moddable-OpenSource/moddable/blob/public/modules/network/websocket/websocket.js#L28

phoddie, 2019-02-15:
> This is where I saw the error

Each link provided in above is to an `import` statement not a `require` as stated above. 

As noted previous, the `import` statement is resolved at build time for modules which are imported during the preload phase of the build and at runtime for the others. The `main.js` file is not preloaded by most manifests, so its `import` statements are evaluated at runtime. In general the contents of $MODDABLE/modules/ are loaded at preload time and so their `import` statements are  evaluated at runtime.

In any case, the original issue reported is a consequence of the network modules being unavailable for the Linux target. This is not a bug.

lll000111, 2019-02-15:
No difference since this isn't a JS runtime, and even then it's semantics. I used the words interchangeably. Lack of discipline, sure, but without actual consequence for what I said.

> This is not a bug.

I never said that it is. Did you read the comment you replied to?

phoddie, 2019-02-15:
> ...this isn't a JS runtime...

If by "this" you are referring to the Moddable SDK, I disagree. XS is a JavaScript engine that conforms to the ECMAScript language specification and therefore provides an environment for executing scripts. That is, by definition, a JavaScript runtime.

> ...even then it's semantics. I used the words interchangeably...

The `require` function and the `import` statement are not fully equivalent, and hence cannot be used interchangeably. There are differences. If there were not, Node.js would have fully embraced the `import` statement in its current form. There remain active conversations in TC39 to adapt JavaScript modules, including the `import` statement for use by Node.js.

The use of `require` in the original comment made it difficult to evaluate the problem being reported, as `require` is used only occasionally in the Moddable SDK runtime.

>> This is not a bug.

> I never said that it is.

The title of this issue is "Cannot build examples". That could reasonably be interpreted as the report of a potential error in the SDK. In closing out this issue, It seemed appropriate, to note the cause of the issue reported.

lll000111, 2019-02-16:
> ...1st paragraph...

Irrelevant to the discussion.

> The `require` function and the `import` statement are not fully equivalent,

I know very well what they do, but the point was one file in calling functions from another file so which one was actually used did not matter. I write `import` and I have Babel translate it to `require` or whatever - as do most people. As long as it's static imports and not dynamic ones, which require can do, so what.

My continuing comments after the issue was closed where mere observations, and that is what I stated:

> By the way, this is all an aside at this point, for me, but interesting to know.