Moddable-OpenSource/moddable #128: Building examples/qrcode for esp32

lll000111, 2019-02-13:
Build platform: Linux

esp-idf installed in ~/esp (and a symlink ~/esp32 => ~/esp because when I installed to ~/esp as instructed by esp-idf installation instructions it could not find it, and when I moved it to ~/esp32 it turned out that during different build steps _both_ paths ~/esp as well as ~/esp32 were used)

Only copying the last lines of output:

```
App "xs_esp32" version: dc94381c
Building component project variables list /home/mha/Projects/moddable/build/tmp/esp32/debug/idf/app_update/component_project_vars.mk
including /home/mha/esp32/esp-idf/components/app_update/Makefile.projbuild...
including /home/mha/esp32/esp-idf/components/bootloader/Makefile.projbuild...
including /home/mha/esp32/esp-idf/components/bootloader_support/Makefile.projbuild...
including /home/mha/esp32/esp-idf/components/coap/Makefile.projbuild...
including /home/mha/esp32/esp-idf/components/esp32/Makefile.projbuild...
including /home/mha/esp32/esp-idf/components/esptool_py/Makefile.projbuild...
including /home/mha/esp32/esp-idf/components/mbedtls/Makefile.projbuild...
including /home/mha/esp32/esp-idf/components/nghttp/Makefile.projbuild...
including /home/mha/esp32/esp-idf/components/partition_table/Makefile.projbuild...
including /home/mha/esp32/esp-idf/components/ulp/Makefile.projbuild...
including /home/mha/esp32/esp-idf/components/unity/Makefile.projbuild...
# cc xsHost.c (strings in flash)
In file included from /home/mha/Projects/moddable/xs/sources/xsCommon.h:41:0,
                 from /home/mha/Projects/moddable/xs/sources/xsAll.h:41,
                 from /home/mha/Projects/moddable/xs/platforms/esp/xsHost.c:38:
/home/mha/Projects/moddable/xs/platforms/esp/xsPlatform.h:102:24: fatal error: lwip/netdb.h: No such file or directory
compilation terminated.
/home/mha/Projects/moddable/build/tmp/esp32/debug/qrcode/makefile:444: recipe for target '/home/mha/Projects/moddable/build/tmp/esp32/debug/lib/xsHost.c.o' failed
make: *** [/home/mha/Projects/moddable/build/tmp/esp32/debug/lib/xsHost.c.o] Error 1
```

barbiani, 2019-02-13:
I saw the same when using the wrong version of idf.

lll000111, 2019-02-13:
What is the "wrong version"? I have 64 bit Linux and `xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0.tar.gz` and I cloned git@github.com:espressif/esp-idf.git

barbiani, 2019-02-13:
It would be 3.1.

git checkout release/v3.1
git submodule update

lll000111, 2019-02-13:
Okay, I could build it now.

closed 2019-02-13 by lll000111

phoddie, 2019-02-13:
@barbiani, thank you!