Moddable-OpenSource/moddable #130: It's not a bug, just a suggestion about: moddable zero

typ7777777, 2019-02-19:
Sorry my bad English, just a suggestion: If **moddable zero** support 720p(or more) led touch monitor and AD it that can just powered by e.g. 25w Portable Solar Charger if you did (search key: e.g. Hanergy 25W 5V Solar Charger) , I think I will buy one & your solar charger(assemble it is easy)! and I think it will sell well in many country, then moddable zero will famous like raspberry pi zero!

phoddie, 2019-02-19:
Thank you for your idea. Unfortunately, the microcontroller used in Moddable Zero is not capable of supporting a 720p display. It is already quite a challenge to drive QVGA display, and 720p has 12x more pixels.

closed 2019-02-19 by phoddie

typ7777777, 2019-02-20:
Thank you spend time to reply! I suggest why not use smartphone tech in chip, you can't change the microcontroller feature that designed by factory, but you can choose any other chips in smartphone field: high integration cheep chip solution for smartphone is exist so people in poor country can buy one. if **moddable zero** 's feature seem like a cheap cell phone(maybe remove Cellularbaseband) but can connect a high resolution LED monitor and can use a Portable Solar Charger as power! you can make big success, and high integration cheep chip desire you to study(e.g. OMAP5432) to assemble moddable zero, China has many company can do it, but their engineer is poor maybe because English doc is hard for them.  
And finally thank your XS, it's so cool, I think your hardware field people can did it also, so I wish your org moddable will long live.  even I'm writing a new language and avoid read your code & tech, but I will run your XS in my language, thanks

phoddie, 2019-02-20:
Thank you for your enthusiasm and kind words. They are very much appreciated.

Our focus for the XS JavaScript engine is microcontrollers. The CPUs found in mobile phones are quite capable of running the high performance JavaScript engines, including SpiderMonkey and V8, found in web browsers.

typ7777777, 2019-02-21:
Yes, I'm try to develop a high performance language support VM also, If I did, I will try to compile your XS to compare how faster if not or how small or if not or how powerful to build any your dream personal custom computer language. If I did and meet your request(I will send one bin.o copy to invite your test the new XS build on my language) and watch times if no one outside to help you improve XS, I suggest just close source, e.g. people like me did not to read your code & tech for protect your thought.  I want to you add & remove XS to make JS better to even build OS (remove something and support byte/short/new(address)...), because originally JS get success by close source and not open it, we can fellow this way. 
finally, thank your reply, If I finish my langue I will talk to you again.
and why do it? some reason is V8/SpiderMonkey I can't support: too heavy and use ASM, and no important peaple care about a no famous people's words.

phoddie, 2019-02-25:
That is a **very** ambitious goal. Good luck!

> to make JS better to even build OS (remove something and support byte/short/new(address)...),

FWIW - you may want to look at `TypedArray` and `DataView` which provide good basic support for working with binary data. We use them often in our projects. `SharedArrayBuffer` is more-or-less the equivalent of `malloc` in the [XS implementation](https://github.com/Moddable-OpenSource/moddable/blob/70e4c2107501302646c04cad073f3b2d36253813/xs/sources/xsAtomics.c#L593).

typ7777777, 2019-02-27:
Thanks your reply, I think use new(address) is more directly than Js' things, because build OS means to write & read kernel memory e.g. graph buffer that address is fixed normally, so just make js more like C++, then can did it(to build OS),
Add I don't want to waste your time to reply me, when I finish a XS Engine based my lang demo, I will info you, Thanks.

typ7777777, 2019-02-27:
Forgot one thing: keep watch KaiOS(formerly firefox OS) the cheapest KaiOS cellphone is only 10$, if moddable zero choose my suggestion (see up), and without Cellularbaseband(small corp can't offer it), maybe you can make moddable zero only 5$ without monitor.  that all my suggestion, 
And don't need reply, except If you interest KaiOS' way, I know a China web browser base chrome source and it only 8M binary, good at embed to a small device.

sophiabee005, 2019-05-23:
I'm impressed, I have to admit. Seldom do I come across a blog that's both educative and entertaining, and let me tell you, you've hit the nail on the head. The issue is something which not enough folks are speaking intelligently about. I am very happy I found this in my search for something relating to this. <a href="https://top5lists.net/todays-the-best-portable-solar-chargers-to-buy/">Best Portable Solar Charger To Buy For </a>