Moddable-OpenSource/moddable #131: feedback on preload.md

allenwb, 2019-02-21:
I very much liked this writeup and have a couple reactions and suggestions.

### [Module Global State](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/preload.md#module-global-state)

This section surprised me, both because of the terminology and the implications of the approach it is recommending.  In particular:

> Sometimes modules need to maintain information for their entire lifetime, independent of a given class instance. These properties are created when the module executes....

>When this module is preloaded, the value of the count property is frozen as part of the module's closure. An exception is thrown on any attempt to modify it. 

ES6+ top-level (ie "global") module declarations were intentionally not given the semantics of properties but instead, for local access, have a semantics that is more similar to function-level lexically scoped variable bindings. This was done to provide both a cleaner conceptual semantics and more implementation flexibility.  By calling them "properties" you are muddying that semantics and potentially causing user confusion.  Also, it suggests that you may not be taking advantage of the available implementation level flexibility.  Perhaps in your implementation you are using a common mechanism to represent both classic global properties and module "globals"? That is fine, but if so you should keep that detail at the implementation level and not expose it users.

What I had expected to see was a statement something like:

> By using `let` to declare the module global `count` the code is explicitly stating that`count` is a mutable variable that must be allocated in RAM.  If we want to define a module global that is immutable and hence a candidate for ROM allocation it should be declared using `const`.  Such `const` declared module "variables" are automatically allocated in ROM if their value can be predetermined during preload processing.  Generally, `const` should be used to declare module globals except for those that are going to be actually used as mutable variables.

finally, the explanation you give in this section seems to be exactly what you would want to say about class static properties.  For example:

```js
class CountingLog {
	log(msg) {
		trace(`${count++}: ${msg}\n`);
	}
}
CountingLog.count = 1;   //or alternatively, using a proposed class static field. 
Object.freeze(CountingLog};
Object.freeze(CountingLog.prototype);

export default CountingLog;
```
In that case using the state object pattern to represent mutable "static" class statemight be appropriate in some cases.

### Making immutable classes easier
I really admire how far you've accomplished  with ROMable XS while maintaining quite strict ES compatibility.  But the pattens for declaring frozen classes are verbose and hence not very user friendly. Perhaps "decorators" could solve the  usability problem, but they would also introduce runtime dependencies that I suspect you would prefer to  not have to deal with in your preload processing

I wonder if you have considered some simpler way to "declare" ROMable (ie, frozen) classes that still would be source code compatible  with more conventional JS engines. Here a a couple ideas the I immediately thought of.  

#### Use an extends function to indicate that a class is immutable

```js
import ROM from "ROMclasses";
class example1 extends ROM() {...};  //inherits as if no extends clause
class example2 extends ROM(anotherClass){...};
//processed by preloader roulghly as:
//const $example1= class example1 {...}
//Object.freeze($example1);
//Oblect.free($example1.prototype);
//const example1 = $example1;
```

The version of "ROMClasses" for use on conventional JS engines would act as a no-op that just returns its argument (or null if no argument).  (assuming a correct fix to the ES `extends null` bug)

#### Use a custom directive

You apparently already place all class objects, prototypes, and methods into ROM.  So we are really only talking about  saving the space of the entries in the mutable object table.  Why not recognize a custom module level directive that essentially says make all classes declared in this module immutable:

```js
"use immutable classes";
class example1 {...}
export class example2{...}
//both example 1 and example2 are frozen and given const bindings
```
Such a directive would be ignored in a conventional JS engine.  You might also consider a "use immutable functions" directive that does something similar for `function` declarations.

I think avoiding a lot of implementation specific directives is generally a good practice but I think they make sense for constrained environments such yours.

phoddie, 2019-02-23:
@allenwb, thank you **very** much for your encouraging words and thoughtful feedback.

> ES6+ top-level (ie "global") module declarations were intentionally not given the semantics of properties but instead, for local access, have a semantics that is more similar to function-level [...]

I was imprecise in the text, as you quickly noted. In XS, these variables are part of the closure of the module's body, as the standard expects. They are not implemented as global properties.

As an aside, module code can use a global property with a symbol to maintain mutable state, as symbols created during preload carry through to runtime (not noted in the initial version of the document):

	const countSymbol = Symbol("count");
	global[countSymbol] = 0;

	class CountingLog {
		log(msg) {
			trace(`${global[countSymbol]++}: ${msg}\n`);
		}
	}

> What I had expected to see was a statement something like:

This is correct, except... the XS preload mechanism has a limitation (bug...) today that treats mutable variables contained in the closure of the module's body if they were declared with `const`. Hence, a workaround is needed. We intend to revisit this limitation when working on private class fields.

```js
CountingLog.count = 1;   //or alternatively, using a proposed class static field. 
Object.freeze(CountingLog};
Object.freeze(CountingLog.prototype);
```

> In that case using the state object pattern to represent mutable "static" class statemight be appropriate in some cases.

The above cannot work because `count` cannot be changed once `CountingLog` is frozen. Also, as the document notes, it is not necessary to explicitly freeze the constructor as the preload implementation freezes all preloaded functions.

> ...the pattens for declaring frozen classes are verbose and hence not very user friendly...
> ...I wonder if you have considered some simpler way to "declare" ROMable (ie, frozen) classes that still would be source code compatible with more conventional JS engines. Here a a couple ideas the I immediately thought of.

We begin by trying to implement the functionality we need using the tools provided in the standard. If that becomes unwieldy, we look for light solutions. The small change to `Object.freeze` to provide recursive freezing here is an example of that. Still, it is not terribly user friendly as you observed. 

> Use an extends function to indicate that a class is immutable

This is an interesting idea with the nice property that it doesn't modify the language. It would be a fair amount of work for us to implement because our native classes currently cannot use `extends`, although that would not impact most user classes.

> Use a custom directive

That would work, and has the nice property of being effectively invisible to other engines. 

Our freezing of objects to store them in flash/ROM is similar to what @erights at Agoric is doing with [`harden`](https://github.com/Agoric/Harden/). I hope we arrive at a single solution that works well for our needs and theirs.

Finally, I will post an update to the document the coming days that incorporates corrections and additions based on your feedback.

allenwb, 2019-02-23:
Of source, I should have said:
'''js
CountingLog.count = 1;   //or alternatively, using a proposed class static field. 
Object.seal(CountingLog};
Object.freeze(CountingLog.prototype);
``` 

For defining a class with a mutable static property.

During ES6 development Mark Miller did briefly suggest `const` function declarations of the form:
```js
const function foo() {...};
```

Might still be a good idea and similarly applicable for class definitions:
```js
const class Klass extends s {...} 
```

`const` is not the ideal word for this and ideally it should be the default rather than having to be explicit. But too much legacy to do that. An experimental implementation in XS might provide some the push necessary to revisit that idea in TC39. 

>Also, as the document notes, it is not necessary to explicitly freeze the constructor as the preload implementation freezes all preloaded functions.

Are there any functions that don't get preloaded?

phoddie, 2019-02-24:
The proposed use of `const` to modify classes and functions is clever. For classes, it is certainly an improvement over a separate call to `Object.freeze` after the class declaration.

> ...and ideally it should be the default rather than having to be explicit.

For classes, it is concise and explicit. For functions outside classes, it could be easy to forget.

> Are there any functions that don't get preloaded?

The XS linker freezes all functions, including built-ins, after loading all modules marked for preload. The behavior is convenient today but can be changed, of course.

phoddie, 2019-07-19:
Today's push to XS solves one of the issues discussed above:

> the XS preload mechanism has a limitation (bug...) today that treats mutable variables contained in the closure of the module's body if they were declared with const. Hence, a workaround is needed. We intend to revisit this limitation when working on private class fields.

With the push, top level declarations in a module are writable through the aliasing mechanism unless they are declared with `const`.

closed 2019-07-27 by phoddie