Moddable-OpenSource/moddable #134: ESP32 does not support 64-bit atomics

jamieparkinson, 2019-03-01:
Hi,

The recent addition of `BigInt64` support is great, but unfortunately this combined with `Atomics` throws some linker errors which I believe are due to the lack of support for 64-bit atomics on the xtensa/esp32 architecture.

**Steps to reproduce**
- Perform an `Atomic` operation on `BigInt64Array` element:
```
const arr = new BigInt64Array(1);
arr[0] = 1;
Atomics.add(a, 0, 1);
```
- Try to compile/run it:
```
mcconfig -d -m -p esp32
```
- Observe errors at the linking stage:
```
...
xs_esp32.a(xsAtomics.c.o): In function `fx_Atomics_or':
xs/sources/xsAtomics.c:180: undefined reference to `__atomic_fetch_add_8'
...
```

The `_8` (8 for 8 bytes in a 64-bit type) suffix is generated by GCC at compile time to be resolved later on by a user-provided implementation: I wonder if providing this in one of the platform files is the right way to go? Even if the implementation isn't truly atomic.

I do think I've understood this issue correctly but would be thrilled to hear that in fact I'm missing something - thank you!

phoddie, 2019-03-01:
@jamieparkinson, I think you have understood the problem well. We, @Moddable-OpenSource,  have exercised `BigInt` quite a bit on ESP32 and ESP8266. We don't use Atomics in our own work, and clearly didn't try the 64-bit operations. Good catch.

I'm uncomfortable with a solution that is knowingly not atomic, unless all it does is `throw`. It seems possible to use a FreeRTOS critical section (e.g. `vTaskEnterCritical`) to get a correct, if inefficient, early implementation. We'll take a look.

jamieparkinson, 2019-03-02:
Thank you for the confirmation! And agreed about the potential implementation. 

I wasn't thinking about this when I wrote the issue but now that I consider it, I assume that now all `Atomics` are broken (since the addition of the 64-bit Atomic to `xsAtomics.c`) due to the nature of the linker error. Will be able to confirm that next week though.

phoddie, 2019-03-02:
No, not all `Atomics` are broken on ESP32. The following builds for ESP32 and generates the expected output of `1`:

```
const buffer = new SharedArrayBuffer(8);
let a = new Uint8Array(buffer);
Atomics.add(a, 0, 1);
trace(a[0], "\n");
```

The reason it works is that the `strip` feature of the `xsl`, the XS Linker, is enabled by default. It determines that `BigInt` isn't used by the project, so omits the `BigInt` support in `Atomics` side-stepping the link error.

phoddie, 2019-03-03:
FWIW - I've tested a fix for this. Should be pushed some time this week.

phoddie, 2019-03-11:
Fix pushed,

closed 2019-03-11 by phoddie