Moddable-OpenSource/moddable #135: manifest.json - usage question

lll000111, 2019-03-01:
I don't quite find an answer in https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/tools/manifest.md

I have JS code with a "main.js" file including a lot of stuff in various sub directories:

**Left:** string used to import the module : **Right:** Path to the files

    "*": "./DEVICE/*",
    "one.core/lib/*": "./DEVICE/node_modules/one.core/lib/*",
    "one.core/lib/system/*": "./DEVICE/node_modules/one.core/lib/system/*",
    "one.core/lib/util/*": "./DEVICE/node_modules/one.core/lib/util/*",

I tried writing the "modules" section of manifest.json like that, I'm not sure what the meaning of the left hand side and right hand side actually is after reading the documentation. Doesn't work, apparently it means something different from what I tried.

bfriedkin, 2019-03-04:
Hello -

We apologize for the confusion here. Indeed it can sometimes be tricky to get the `modules` properties and associated application `import` statements correct. Your understanding is essentially correct. The `modules` section defines which modules are built by the application. Entries in a modules array that have named keys are matched by import statements in the app. The property values for these entries do correspond to path names to the JavaScript modules source. Wildcards can be used in both the keys and the values. Based on your notes, I believe that the issues you are having are due to small syntax errors in your `main.js` and `manifest.json` files.

Note that we don't use file extensions, e.g. ".js" in import statements or in the manifest.

There are a number of manifest files in the [examples](https://github.com/Moddable-OpenSource/moddable/tree/public/examples) directory that use wildcards and named keys. In particular the [`manifest_commodetto.json`](https://github.com/Moddable-OpenSource/moddable/blob/public/examples/manifest_commodetto.json) and [`manifest_piu.json`](https://github.com/Moddable-OpenSource/moddable/blob/public/examples/manifest_piu.json) files likely include some patterns that cover what you are trying to achieve.

That said, a simple and focused working example can often help to resolve issues at hand. I've attached an `imports` example app the shows a handful of different module and import variations. The example is based on the directory hierarchy you've shared here.

If the working example doesn't help to get you past the issue at hand, please provide more details about the error(s) you are seeing at build/run time.

Regards,
Brian

[imports.zip](https://github.com/Moddable-OpenSource/moddable/files/2923938/imports.zip)

lll000111, 2019-03-04:
Thank you!

> Note that we don't use file extensions, e.g. ".js" in import statements or in the manifest.

Feature request: Please allow the file extension. Our project shares code across browser, React Native, node.js, and use of the extension in import statements is a Best Practice. Some bundlers even require it. Some platforms won't find it without extension.

Examples:

https://github.com/Microsoft/TypeScript/issues/16577

https://stackoverflow.com/questions/44481851/does-es6-import-export-need-js-extension

lll000111, 2019-03-04:
I played around with your example zip. When I change an import in main.js to be nonsensical (random string pointing nowhere) I get no errors and the build just runs through.

I think there should definitely be an error (and a message).

lll000111, 2019-03-04:
Update 2: When I remove all "modules" entries but the first one for main than the build runs through and the simulator starts (building for Linux during these basic build tests). The runtime then complains about not finding the module (of course).

lll000111, 2019-03-04:
The two lines

```
"one.core/lib/system/*": "./node_modules/one.core/lib/system/*",
"one.core/lib/util/*": "./node_modules/one.core/lib/util/*",
```

cause the error. I don't know why. The directories exist.

lll000111, 2019-03-05:
Runtime error in xsbug: `require: module "./util/recipe-find-linked-rules" not found!`

But there is a line in manifest.json for this:

    "./util/*": "./node_modules/one.core/lib/util/*",

Run from the build root (same directory as manifest.json and where I run mcconfig from):

```
$ ls -l ./node_modules/one.core/lib/util/
total 48
...
-rw-r--r-- 1 mha mha  1252 Mar  4 13:00 recipe-find-linked-rules.js
...
```

bfriedkin, 2019-03-05:
Hello -

Unfortunately it is difficult to know the cause of this error without knowing what the `import` statement is.

I've attached another simple working example here that uses your same `./util/*` pattern. Note that in this example, using `./util/*` or `util/*` in the `manifest.json` and `main.js` files achieve the same result.

Regards,
Brian

[imports2.zip](https://github.com/Moddable-OpenSource/moddable/files/2931716/imports2.zip)

closed 2019-03-05 by bfriedkin

lll000111, 2019-03-05:
> Unfortunately it is difficult to know the cause of this error without knowing what the import statement is.

I'm working on a test zip archive...

lll000111, 2019-03-05:
By the way, doesn't the error message I posted above show the path?

> `require: module "./util/recipe-find-linked-rules" not found!`

That file is includes twice. Once from a file in the same directory that uses `./recipe-find-linked-rules`, and once from a file in the parent directory which uses the path that is shown.

lll000111, 2019-03-06:
Never mind, I decided to avoid the whole issue by changing the paths during the build process.