Moddable-OpenSource/moddable #136: Trouble with "export async function"?

lll000111, 2019-03-04:
### Output

I'll just show one file, there are several, the location always is an `export async function` line.

```
mha@mint-virtual-machine ~/Projects/one.moddable/DEVICE $ mcconfig -d -m -p lin
# xsc module-importer.xsb
# xsc instance.xsb
/home/mha/Projects/one.moddable/DEVICE/node_modules/one.core/lib/module-importer.js:34: error: invalid export identifier!
/home/mha/Projects/one.moddable/DEVICE/node_modules/one.core/lib/module-importer.js:42: error: missing ;!
/home/mha/Projects/one.moddable/DEVICE/node_modules/one.core/lib/module-importer.js:62: error: missing ;!
### 3 error(s)
# xsc microdata-to-id-hash.xsb
/home/mha/Projects/moddable/build/tmp/lin/debug/mc/DEVICE/makefile:315: recipe for target '/home/mha/Projects/moddable/build/tmp/lin/debug/mc/DEVICE/modules/module-importer.xsb' failed
make: *** [/home/mha/Projects/moddable/build/tmp/lin/debug/mc/DEVICE/modules/module-importer.xsb] Error 22
make: *** Waiting for unfinished jobs....
...
```

Code in module-importer, starting from line 34:

```js
export async function createObjects(WriteStorage, {
  moduleName,
  version = '1.0',
  code
}) {
  let clobResult;

  try {
    clobResult = await WriteStorage.storeUTF8Clob(code);
  } catch (err) {
    err.stack += new Error().stack;
    throw err;
  }

  if (clobResult.status === CREATION_STATUS.EXISTS) {
    MessageBus.send('log', `Code CLOB ${clobResult.hash} already exists`);
  }

  const module = {
    type: 'Module',
    name: '@module/' + moduleName.replace(/\.js$/, ''),
    provides: [],
    requires: extractRequires(code),
    version,
    code: clobResult.hash
  };

  try {
    return await WriteStorage.storeVersionedObject(module);
  } catch (err) {
    err.stack += new Error().stack;
    throw err;
  }
}
```

lll000111, 2019-03-05:
It indeed looks like this particular syntax causes problems.

Workaround: I removed the `export` from those functions and do an extra `export {theFunction};` at the end of the file. Those errors have stopped showing up.

phoddie, 2019-03-07:
Thank you for the detailed report. A fix will be available in the coming days.

lll000111, 2019-03-11:
Fixed in https://github.com/Moddable-OpenSource/moddable/commit/5f9235280af8eb532127569ba2c9723f87e543e3

closed 2019-03-11 by lll000111