Moddable-OpenSource/moddable #139: Error compiling "cryptdigest" example

lll000111, 2019-03-07:
Platform: Linux

```
# cc xsBigIntEx.c.o
# cc mont.c.o
# cc ec.c.o
# cc modular.c.o
# cc bin.c.o
# cc modBase64.c.o
# cc mc.resources.c
/home/mha/Projects/moddable/modules/crypt/arith/xsBigIntEx.c: In function ‘fxBigInt_ec_double’:
/home/mha/Projects/moddable/modules/crypt/arith/xsBigIntEx.c:544:17: error: ‘true’ undeclared (first use in this function)
   r->identity = true;
                 ^
/home/mha/Projects/moddable/modules/crypt/arith/xsBigIntEx.c:544:17: note: each undeclared identifier is reported only once for each function it appears in
/home/mha/Projects/moddable/modules/crypt/arith/xsBigIntEx.c:548:16: error: ‘false’ undeclared (first use in this function)
  r->identity = false;
                ^
/home/mha/Projects/moddable/modules/crypt/arith/xsBigIntEx.c: In function ‘fxBigInt_ec_add’:
/home/mha/Projects/moddable/modules/crypt/arith/xsBigIntEx.c:626:18: error: ‘true’ undeclared (first use in this function)
    r->identity = true;
                  ^
/home/mha/Projects/moddable/modules/crypt/arith/xsBigIntEx.c:631:16: error: ‘false’ undeclared (first use in this function)
  r->identity = false;
                ^
/home/mha/Projects/moddable/modules/crypt/arith/xsBigIntEx.c: In function ‘fxBigInt_ec_mul’:
/home/mha/Projects/moddable/modules/crypt/arith/xsBigIntEx.c:678:16: error: ‘true’ undeclared (first use in this function)
  r->identity = true;
                ^
/home/mha/Projects/moddable/modules/crypt/arith/xsBigIntEx.c: In function ‘fxBigInt_ec_mul2’:
/home/mha/Projects/moddable/modules/crypt/arith/xsBigIntEx.c:698:16: error: ‘true’ undeclared (first use in this function)
  r->identity = true;
                ^
/home/mha/Projects/moddable/build/tmp/lin/debug/mc/cryptdigest/makefile:476: recipe for target '/home/mha/Projects/moddable/build/tmp/lin/debug/mc/cryptdigest/xsBigIntEx.c.o' failed
make: *** [/home/mha/Projects/moddable/build/tmp/lin/debug/mc/cryptdigest/xsBigIntEx.c.o] Error 1
```

<br>

Unfortunately, when I define true and false the compilation *appears* to run through...

```
$ mcconfig -d -m -p lin
# xsid xsBigIntEx.c.xsi
# xsl modules
# cc mc.xs.c
# cc xsBigIntEx.c.o
# ld mc.so
mha@mint-virtual-machine ~/Projects/one.moddable/DEVICE $ 
```

... but nothing happens (xsbug is open and remains silent).

bfriedkin, 2019-03-07:
Hello -

Thank you for reporting this issue. We have been able to reproduce the error building on Ubuntu 18.04 Linux as well.

An easy workaround is to add the following to xsBigIntEx.c:

```
#ifndef MIN
#define MIN(a, b)	((a) <= (b) ? (a) : (b))
#endif
#ifndef true
	#define true (1)
#endif
#ifndef false
	#define false (0)
#endif
```
Keep in mind that on desktop hosts, you need to manually launch `xsbug` before building and running your Moddable app. When I do that I see the following in the `xsbug` console:

```
Calculated hash: s3pPLMBiTxaQ9kYGzzhZRbK+xOo=
Expected hash: s3pPLMBiTxaQ9kYGzzhZRbK+xOo=
PASS
PASS
PASS
```

Would you be able to give this another try? Let us know how that goes. We will review the fix and roll into our next open source release.

Regards,
Brian

lll000111, 2019-03-08:
I had done all of that, only that I used `#define true 1` without parentheses but I tried exactly your code just now. I had also started xsbug separately, I had already noticed that behavior. The compilation finishes as shown above and xsbug remains completely silent and empty.

lll000111, 2019-03-08:
All "Linux" is stuck now for some reason, meaning the compile runs through and then nothing happens. Even the things that worked just fine. I did not change anything in `$MODDABLE` apart from the above `#define`, `git status` agrees.

lll000111, 2019-03-08:
Looking at the generated makefile and which files where generated during the built, I think `$(BIN_DIR)/mc.so` is not created. I get the output `# ld mc.so` (see above) but the file is nowhere to be found.

closed 2019-03-08 by lll000111