Moddable-OpenSource/moddable #140: "cannot coerce undefined to object" - but the value is a Set

lll000111, 2019-03-08:
I have a weird problem that running in isolation works just fine.

### Code

(Excerpt)

```js
  if (thing.referenceTo !== undefined) {
    trace(thing.referenceTo + '\n');
    for (let entry of thing.referenceTo) {
      trace(entry + '\n');
    }

    trace('\n');
    [...thing.referenceTo];
```

`thing.referenceTo' is a `Set` with a single string value '*' inside.

### Output

_I wish I could copy & paste from the console in xsbug...._

```
[object Set]
'*'

the-file.js (178) # Break: the-function: cannot coerce undefined to object!
```

The same thing happens in another place (the `case` is resched because `Object.prototype.toString.call` returned string "Set", which was extracted, so `obj` is sure to be a `Set`):

```
case 'Set':
      return actualStringify([...obj], seenObjects);
```

All of that is well-tested, the exact same code, just a mini-example of our stuff, runs on node.js browser, react-native - and even low.js (on [this board](https://www.neonious.com/neoniousOne)).

### Works when run in isolation

I tried to run this as an isolated example:

```js
function isString(thing) {
  return typeof thing === 'string';
}

const thing = {
  referenceTo: new Set(['*'])
};

trace(thing.referenceTo + '\n');

for (let entry of thing.referenceTo) {
  trace(entry + '\n');
}

if (
  thing.referenceTo !== undefined &&
  (
    !(thing.referenceTo instanceof Set) ||
    ![...thing.referenceTo].every(type => isString(type))
  )
) {
  throw new Error('Error');
}
```

but this worked fine.

lll000111, 2019-03-08:
PS: I'll keep debugging and if I can't think of anything else create a runnable example to send you.

lll000111, 2019-03-08:
An update: When I replace the `[...setObject]` by `Array.from(setObj)` it works. So it seems that there is an issue specifically when the object spread syntax is used.

lll000111, 2019-03-09:
## Reproducible example

I ran it on Linux, but I discovered the issue originally on ESP32.

There are two different errors. To discover the first one just uncomment the code marked `Error #1`.

```js
'use strict';

function isString (thing) {
    return typeof thing === 'string';
}

function ensureRecipeRule (thing) {
    if (thing.referenceTo instanceof Set) {
        [...thing.referenceTo].every(type => isString(type));
    }

    // ERROR #1
    // THE EXACT SAME THING AS ABOVE REPEATED: "callback is no function"
    // if (thing.referenceTo instanceof Set) {
    //     [...thing.referenceTo].every(type => isString(type));
    // }

    // ERROR #2
    // "cannot coerce undefined to object"
    if (thing.idReferenceTo instanceof Set) {
        [...thing.idReferenceTo].every(type => isString(type));
    }
}

trace('Start\n');

[
    {
        referenceTo: new Set(['Person']),
    },
    {
        idReferenceTo: new Set(['Group'])
    }
]
.map(rule => ensureRecipeRule(rule));

trace('End\n');
```

Result (Error 2):

[image]

lll000111, 2019-03-10:
Another possibly related example:

_Test performed on Linux._

### Code

```js
const a1 = [1,2,3];
const a2 = [];
const aa = [...a1, ...a2, undefined];

debugger
```

### Expected Result

`[1, 2, 3, undefined]`

### Actual Result

`undefined`

phoddie, 2020-10-14:
This was fixed a while back.

closed 2020-11-17 by phoddie