Moddable-OpenSource/moddable #144: Error objects should be writable

lll000111, 2019-03-10:
I give `Error` objects names (i.e. setting `name`property, sometimes replacing an existing `message` property too) because I prefer this over sub-classing Errors, which only is simple on ES 2015 and with a functional style I don't have a single "class", "this", "bind" in any of my code, it's "just functions". This is perfectly legal JS.

## Code

```js
const e = new Error('test');
e.name = 'TestError';
trace(e.name + ' ' + e.message);
```

## Result

[image]

lll000111, 2019-03-10:
It seems to me that this is due to [xs/sources/xsError.c setting `XS_GET_ONLY`](https://github.com/Moddable-OpenSource/moddable/blob/public/xs/sources/xsError.c#L121).

I set this to `XS_INTERNAL_FLAG | XS_GETTER_FLAG | XS_SETTER_FLAG | XS_DONT_DELETE_FLAG | XS_DONT_ENUM_FLAG` and then I could set a message in JS code.

However, I would also like to add custom properties to the object, that's still some other limitation...

lll000111, 2019-03-11:
## Workaround

_Just for the record_

This works:

```js
const err = new Error("Sum Ting Wong, Wi Tu Lo, Ho Lee Fuk, Bang Ding Ow";
Object.defineProperty(
  err,
  'name',
  {
    enumerable: false,
    writable: true,
    configurable: true,
    value: 'FileNotFoundError'
  }
);
```

phoddie, 2019-03-11:
This is a consequence of freezing of the `Error.prototype` during preload. You can read more about that [here](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/preload.md).

closed 2019-03-11 by phoddie

lll000111, 2019-03-11:
As I wrote, the way this works and is supposed to work, looking at other runtimes, is that a new property IN THE INSTANCE is created. Nobody asks about modifying the prototype object.

lll000111, 2019-03-11:
I read the link again, I had already read it some time ago, just to be sure I did not misread.

The example there explicitly directly changes the prototype object. This has nothing to do with *this* issue here.

phoddie, 2019-03-11:
It is a consequence of `Error.prototype` being frozen. The `name` property is on the prototype as per [section 19.5.3.3](https://tc39.github.io/ecma262/#sec-error.prototype.name) of the ECMAScript language specification. When an object is frozen, writes to existing properties fail, including those on instances that use the frozen prototype.

lll000111, 2019-03-11:
Okay, they indeed do. TIL.

lll000111, 2019-03-11:
Interesting the Object.defineProperty() still works.