Moddable-OpenSource/moddable #145: Support for ESP32 CAM

TonyVie, 2019-03-11:
Hi !

Do you support ESP32-CAM module (from AI Thinker). It features a 2MP camera and 4MBit PSRAM.
(https://github.com/espressif/esp32-camera
https://www.cnx-software.com/2018/09/30/esp32-cam-esp32-camera-board/)

Im planning to build a camera with an embedded thermal printer (via TTL) and Im wondering if moddable is a good way to go :)

Thanks !

phoddie, 2019-03-11:
Thanks for asking. We don't currently support the ESP32-CAM. There is [support for ArduCam](https://github.com/Moddable-OpenSource/moddable/tree/public/modules/drivers/arducam) on ESP8266 in this repository. That works nicely. We'd certainly be happy to add or integrate support for ESP32-CAM.  If you want to contribute the software, that would be great. If you would consider contributing a board, we can look at implementing the driver.

phoddie, 2019-04-22:
It appears this product is no longer available for purchase. For example,

> This product has been discontinued at Seeed Studio

Happy to look at supporting a camera product that is available in the future. Closing this request.

closed 2019-04-22 by phoddie