Moddable-OpenSource/moddable #146: ESP32: How do I interpret this crash?

lll000111, 2019-03-11:
```
Flashing binaries to serial port /dev/ttyUSB1 (app at offset 0x10000 )...
esptool.py v2.6
Serial port /dev/ttyUSB1
Connecting.....
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, BLK3 partially reserved, Coding Scheme 3/4
MAC: 30:ae:a4:cc:3f:9c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 13568 bytes to 8907...
Wrote 13568 bytes (8907 compressed) at 0x00001000 in 0.1 seconds (effective 968.0 kbit/s)...
Hash of data verified.
Compressed 780096 bytes to 453304...
Wrote 780096 bytes (453304 compressed) at 0x00010000 in 6.4 seconds (effective 978.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 122...
Wrote 3072 bytes (122 compressed) at 0x00008000 in 0.0 seconds (effective 1472.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
�x
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d238c  PS      : 0x00060f30  A0      : 0x800d1a90  A1      : 0x3ffb7590  
A2      : 0x3ffb24ec  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x3f40f400  A7      : 0x00000000  A8      : 0x800d2377  A9      : 0x3ffb7570  
A10     : 0x3f40f480  A11     : 0x3f404c21  A12     : 0x0000002c  A13     : 0x3ffb76ec  
A14     : 0x00080000  A15     : 0x00000011  SAR     : 0x00000014  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000001c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
Backtrace: 0x400d238c:0x3ffb7590 0x400d1a8d:0x3ffb7720 0x400d1a97:0x3ffb7770
Entering gdb stub now.
```

Nothing happens after the last line.

lll000111, 2019-03-11:
**Next question:** I built without `-d`, now I get

```
Hard resetting via RTS pin...
MONITOR
--- idf_monitor on /dev/ttyUSB1 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:644
load:0x40078000,len:6208
load:0x40080000,len:5156
0x40080000: _iram_start at /home/mha/esp32/esp-idf/components/freertos/xtensa_vectors.S:1685

entry 0x40080264
0x40080264: _Level5Vector at ??:?
```

and then... silence. Is this a crash? Must be, nothing else happens after this text.

phoddie, 2019-03-11:
You have crashed in native code. Native debugging is out-of-scope for this project. Please refer to the ESP-IDF project.

closed 2019-03-11 by phoddie

lll000111, 2019-03-11:
But the crash is related to manifest.json settings and that's THIS project.

I get the crash when I increase manifest.json creation.static just a bit.

I must admit I have no clue what the important values are and what they mean. What exactly is static anyway? I did read Xs in C and still don't know.

I need to give a bit more to my app then what the tiny sample apps get. I have a 4MB RAM WROVER-B-KIT ESP32 so that should be no problem. I get "could not allocate slot" errors early on already.

phoddie, 2019-03-11:
It does seem like you are swimming in stronger currents than you are prepared to handle. That must be frustrating. There's certainly not enough detail in this report to be able to offer assistance. That's also frustrating.

lll000111, 2019-03-11:
Well, if I knew what things like "static" means..

I'm unused to YOUR environment, true (but find it quite readable but of course it's a lot of code), and I have not done much C &mdash; but when I did I contributed to Linux kernel code (kernel 2.4, network, firewall and NAT). So I'm not some new student completely in over his head. I think some memory basics should be part of the documentation and I should not have to scan all the XS source code to find out. Especially when there are some tricky issues like this: there is a separate esp and esp32 directory, but the only place where I find `staticSize` being used is in an esp/ directory, and there is no esp32 directory. So... do I interpret this as that there is no code using this for esp32, or that this file in this context is for both platforms? Yes I could scan the rest of the code but things like manifest.json entries should IMHO just be documented.