Moddable-OpenSource/moddable #147: manifest.json: "creation" values are not explained

lll000111, 2019-03-11:
https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/tools/manifest.md#creation

The only explanation:

> The creation object defines the creation parameters of the XS machine that will run the app. See [the XS in C documentation](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/XS%20in%20C.md) for details.

That page does not tell me e.g. what `static` does/means. It is briefly mentioned [here](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/XS%20in%20C.md#machine-allocation) as part of a `struct`. In the example below exactly this value is left out.

In particular, is there any number that sets a maximum? Or is memory allocated as long as there is any free physical memory left? Is this "`static`" property the maximum amount of memory used by a virtual machine, in bytes?

<br>
*Related: #146 *

lll000111, 2019-03-13:
Whatever. I'll figure it out.

closed 2019-03-13 by lll000111

lll000111, 2019-03-13:
For the record, should anyone get here, because unanswered questions that you come across while searching for a problem you have have an xkcd and also a Dilbert, I think:

Found this:

https://github.com/Moddable-OpenSource/moddable/blob/public/xs/platforms/esp/xsHost.c#L660

which defines `ESP_cloneMachine` which uses `stackSize`  for a call to `malloc`.

That function is used by `./build/devices/esp32/xsProj/main/main.c` to create the virtual machine:

https://github.com/Moddable-OpenSource/moddable/blob/public/build/devices/esp32/xsProj/main/main.c#L141

chances, 2020-12-21:
The [machine allocation documentation](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/XS%20in%20C.md#machine-allocation) in the XS in C document _still_ does not explain what the `staticSize` (i.e. `static` in manifests) is/does.

@lll000111 Could you reopen this issue?