Moddable-OpenSource/moddable #148: mcconfig error

dora-gt, 2019-03-15:
Hi,

When I try to `mcconfig -d -m -p mac` for my project, the following error occurs.

```
$ mcconfig -d -m -p mac
# xsc main.xsb
/Users/xxx/src/main.js.map:5: error: source map: version is no integer!
```

`main.js.map` is like the below.

```js:main.js.map
{"version":3,"file":"main.js","sourceRoot":"","sources":["main.ts"],"names":[],"mappings":"AAAA,IAAO,IAAI,CAEV;AAFD,WAAO,IAAI;IACP,KAAK,CAAC,WAAW,CAAC,CAAC;AACvB,CAAC,EAFM,IAAI,KAAJ,IAAI,QAEV"}
```

It seems that `version` is `3` and it is an Integer. What is the problem?

lll000111, 2019-03-15:
No answer but that looks like it could be easy to find: The code is at

https://github.com/Moddable-OpenSource/moddable/blob/public/xs/sources/xsSourceMap.c#L93

and I would just try to do some debugging from there.

phoddie, 2019-03-15:
Thank you for the report, @dora-gt. Source maps were unintentionally broken as a result of work to improve `test262` conformance. The fix will be included in the next update to this repository, likely next week.

If you would like to try that sooner, modify `xsLexical.c` in two places. Change line [1377](https://github.com/Moddable-OpenSource/moddable/blob/9023a8eb30c52976c287e6adc6031538d85d697a/xs/sources/xsLexical.c#L1377) from:

`      fxGetNextNumberE(parser, 0);`

to

`      fxGetNextNumberE(parser, 1);`

And line [1410](https://github.com/Moddable-OpenSource/moddable/blob/9023a8eb30c52976c287e6adc6031538d85d697a/xs/sources/xsLexical.c#L1410) from:

`      fxGetNextNumberE(parser, 1);`

to

`      fxGetNextNumberE(parser, 0);`

dora-gt, 2019-03-16:
Thank you! It worked!

closed 2019-03-16 by dora-gt

phoddie, 2019-03-16:
Excellent. Thank you for verifying the fix.