Moddable-OpenSource/moddable #151: ESP32 (Espressif): Does the PSRAM problem of ESP-IDF affect Moddable?

lll000111, 2019-03-15:
I found this still open issue while working on using the PSRAM and it seems to me that Moddable is affected too when building for this kind of hardware.

https://github.com/espressif/esp-idf/issues/2892

> We stumbled upon the fact that cache issue with PSRAM still exist, even in the newest development environment. This can produce random crashes, even if the code is 100 % valid.

<br>

As an aside, Neonious [mention of their dlmalloc implementation for ESP-IDF](http://bbs.esp32.com/viewtopic.php?f=13&t=8628&sid=1acc8bd897e72cf450ad9eb71491d732) also looks interesting if their statements are correct.

> The default memory allocator in ESP-IDF is self made by Espressif (at least so it seems). It is not very fast, and becomes very slow when memory gets fragmented. This problem becomes evident when using SPI RAM.
>
> esp-idf-dlmalloc is a fork of ESP-IDF which was modified to use dlmalloc, an industry standard memory allocator. It is almost twice as fast as the default memory allocator, and does not slow down notably with fragmented memory.

But if Espressif should pick it up everybody gets it in the end anyway.

lprader, 2019-03-15:
Later in the thread you linked to it's mentioned that the PSRAM problem is only an issue in single core mode (see [this comment](https://github.com/espressif/esp-idf/issues/2892#issuecomment-459687245)).

I'm not sure what the "PSRAM problem" is since I didn't find a link to the original issue, but the Moddable SDK uses single core mode by default. You can use multi core mode, but I assume whatever the PSRAM problem is would affect you if you do.

lll000111, 2019-03-15:
> it's mentioned that the PSRAM problem is only an issue in single core mode

@lprader You read that wrong &mdash; It is the opposite :-)

> I'm not sure what the "PSRAM problem" is 

The linked thread talks about it, as well as the linked ESP forum post? Okay, the forum post is linked in the linked thread by an Espressif employee. There are crashes and data corruption.

lprader, 2019-03-15:
Ha, actually I read it correctly, but worded my first sentence incorrectly. My bad. I meant "it's mentioned that the PSRAM problem is **not** an issue in single core mode."

Anyways, like I said: if you switch to using multi core mode I assume it will affect you.

closed 2019-03-21 by lll000111