Moddable-OpenSource/moddable #152: [Question] How to pass Date verification

horihiro, 2019-03-16:
Now [httpsget](https://github.com/Moddable-OpenSource/moddable/tree/public/examples/network/http/httpsget) / [httpsgetjson](https://github.com/Moddable-OpenSource/moddable/tree/public/examples/network/http/httpsgetjson) examples don't work and message 'date validation failed on received certificate' is shown in console.
It seems that it caused by 53945e7 (adding [date vaidation for certificates](https://github.com/Moddable-OpenSource/moddable/blob/public/modules/crypt/ssl/ssl_cert.js#L113)).

In order to avoid the message, is it the only way to set current time by SNTP?

phoddie, 2019-03-17:
@horihiro - yes, recent changes enable date validation for certificates by default. From a security perspective, this is correct. From a development perspective, it can be inconvenient, particularly on microcontroller's that do not have a built-in battery backed real-time clock. 

The `SecureSocket` class constructor's dictionary implements [the optional `verify` property to disable certificate verification](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/network/securesocket.md#dictionary), which includes skipping the date check. For example, you can modify the [httpsget](url
https://github.com/Moddable-OpenSource/moddable/tree/public/examples/network/http/httpsget) example this way:

`````
let request = new Request({host: "www.mozilla.org", path: "/", response: String,
		port: 443, Socket: SecureSocket, secure: {verify: false}});
`````

If you don't mind waiting a second for the time to arrive, you can simply set the SNTP server on the command line:

    mcconfig -d -m -p esp sntp=pool.ntp.org ssid=foo password=bar

Or you can include it in the `config` section of your manifest:

	"config": {
		"sntp": "pool.ntp.org",
	}

horihiro, 2019-03-17:
@phoddie , Thanks for your comment!
I appreciate you so much!!

I confirmed that mcconfig command option `sntp=pool.ntp.org` does work well.
Could you please let me know where this option's explanation is described?
I can't find that in the following document.
https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/tools/tools.md#arguments

Best Regards,

phoddie, 2019-03-17:
@horihiro - The `mcconfig` tool has a general purpose feature on the command line to configure the application being built. Any arguments of the form `name=value`, such as `sntp=pool.ntp.org` or `ssid=Moddable`, are added to the `mc/config` module of the application and is available to scripts. The `mc/config` module source code looks like this:
`````
/* WARNING: This file is automatically generated. Do not edit. */

export default Object.freeze({
	"ssid": "Moddable",
	"sntp": "pool.ntp.org"
});
`````

Scripts access this module by importing as usual:

`````
import config from "mc/config";

new SNTP({host: config.sntp}, function(message, value) {
    ...
`````

The documentation for the [arguments of `mcconfig`](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/tools/tools.md#arguments) describes some of the name/value combinations supported by modules in the SDK (e.g. `ssid`, `password`, `screen`, and `touch`). It does not describe the general mechanism mentioned above,.

horihiro, 2019-03-17:
Thanks for the information for passing configuration!

I tried to work this code.
```
let request = new Request({host: "www.howsmyssl.com", path: "/a/check", response: String,
						  port: 443, Socket: SecureSocket, secure: {verify: false} });
```

But the program doesn't work due to following date check part.
https://github.com/Moddable-OpenSource/moddable/blob/public/modules/crypt/ssl/ssl_cert.js#L208

```
	register(cert) {
		let validity = X509.decodeTBS(X509.decode(new Uint8Array(cert)).tbs).validity, now = Date.now();
		if (!((validity.from < now) && (now < validity.to)))
			throw new Error("date validation failed");

		this.registeredCerts.push(cert);
};
```

Is there any way to avoid checking by this `register` function?

Though I removed `der` resource (`$(MODULES)/crypt/data/ca***`) in a manifest.json, the behavior doesn't change.

horihiro, 2019-03-17:
I found `register` is executed at following line even if `options.scue.verify' is set to `false` always.

https://github.com/Moddable-OpenSource/moddable/blob/public/modules/crypt/ssl/ssl_handshake.js#L505

Is this right behavior?

phoddie, 2019-03-17:
Oops. That takes a code path unanticipated by the previous change. As a workaround, just comment out the verification check in the `register` function above. A proper fix is a little more involved. That'll be in the next push.

horihiro, 2019-03-17:
Thanks @phoddie 
I'm looking forward to the next update.

phoddie, 2019-04-08:
Manifests of examples that require correct date have been updated to include sntp host so time/date is set on launch.

closed 2019-04-08 by phoddie