Moddable-OpenSource/moddable #153: esptool error

ghamaya, 2019-03-17:
(Beginner’s question)
After I updated my project, I had an esptool error.

```
$ mcconfig -d -m -p esp
# ld main.bin
Traceback (most recent call last):
  File "/Users/amaya/esp/ESP8266_RTOS_SDK/components/esptool_py/esptool/esptool.py", line 35, in <module>
    import serial.tools.list_ports as list_ports
ImportError: No module named serial.tools.list_ports
make: *** [/Users/amaya/Projects/moddable/build/bin/esp/debug/helloworld/main.bin] Error 1

```
I suppose my previous project used ~/esp/esptool/esptool
Please guide me what should I do.

barbiani, 2019-03-17:
Did you install the python pre requisites?

https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#get-started-get-packages

ghamaya, 2019-03-17:
Thank you.
I am using ESP8266

ghamaya, 2019-03-17:
Thank you.
I am using ESP8266 SDK. It seems differences from ESP32. I would check Espressif document.

closed 2019-03-19 by ghamaya