Moddable-OpenSource/moddable #157: Ota implementation

diogoviannaaraujo, 2019-03-28:
Hi,

Im trying to implement a module to bridge https_ota to js, but im getting errors when trying to include "esp_https_ota.h", how is the correct way to port Esp-idf libraries in a module?

Ive seen there is some ota implementation inside ESP32 device build folder, but can't find any instructions or examples. Also found that partitions.csv has a ota table commented. Is that functions available?

Thanks

phoddie, 2019-03-28:
Rather than using `http_ota`, use the [OTA module](https://github.com/Moddable-OpenSource/moddable/tree/public/build/devices/esp32/modules/ota) that's already in place for the ESP32. It wraps the core OTA functionality from Espressif.

Here's a simple example:

``````
import OTA from "esp32/ota";
import {Request} from "http"

let ota = new OTA;

let request = new Request({host: "www.example.com", path: "/ota/firmware.bin"});
request.callback = function(message, value, etc) {
	if (4 == message) {
		let bytes = this.read(ArrayBuffer);
		trace(bytes.byteLength, "\n");
		ota.write(bytes);
	}

	if (5 == message) {
		trace(`\n\nTransfer complete.\n`);
		ota.complete();
	}
}

``````

And here's the manifest:

`````
{
	"include": [
		"$(MODDABLE)/examples/manifest_base.json",
		"$(MODDABLE)/examples/manifest_net.json",
	],
	"modules": {
		"*": [
			"./main",
			"$(MODULES)/network/http/*"
		],
		"esp32/ota": "$(MODDABLE)/build/devices/esp32/modules/ota/*"
	},
	"preload": [
		"esp32/ota",
		"http",
	],
}
`````

The Moddable SDK partition is set-up to maximize program space so does not have the partitions required for OTA. You will need to create a `partitions.csv` that supports OTA. The default The Espressif SDK can help with that.

closed 2019-03-28 by phoddie

diogoviannaaraujo, 2019-03-28:
Thanks a lot for the example, phoddie.

How will this handle the download? Will this write directly to the ota partition or it will need to load all in memory first?

mauroForlimpopoli, 2021-03-13:
I want to add OTA to my project.  In the module of manifest:
"esp32/ota": "$(MODDABLE)/build/devices/esp32/modules/ota/*"
and in the platform esp32:
"build": {
	"PARTITIONS_FILE": "$(MODDABLE)/contributed/modClock/sdkconfig/partitions.csv"
}
in the main.js :
let ota = new OTA;
When the debugger stars it gives this error in this line:  let ota = new OTA; 

C:\Users\fly3\projects\moddable\build\devices\esp32\modules\ota\ota.c (54) # Break: OTA: no update parition!

how can I build a correct partition ?

diogoviannaaraujo, 2021-03-13:
Hey you have to create your own partitions file with a ota partition. Check here https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/partition-tables.html

mauroForlimpopoli, 2021-03-14:
Thank you Diogo, but it's too complicated for me :) . By using the Moddable where can I see where mycode resides  and where place  the partition of OTA? can you give me one more help?

diogoviannaaraujo, 2021-03-14:
You need to create a partitions.csv file in your project folder and point that in your manifest where you are pointing to the one in the clock example. Then, in that file you should describe the partitions for you esp32, its line by line separated by commas.
For what partitions to use read that link, is not that complicated, you will get the point, its just a little long.

mauroForlimpopoli, 2021-03-14:
This is the CSV I used: (C:\Users\fly3\projects\moddable\contributed\modClock\sdkconfig)

nvs,      data, nvs,     0x009000, 0x4000,
otadata,  data, ota,     0x00d000, 0x2000,
phy_init, data, phy,     0x00f000, 0x1000,
ota_0,    app,  ota_0,   0x010000, 0x140000,
ota_1,    app,  ota_1,   0x150000, 0x140000,
xs,       0x40, 1,       0x290000, 0x010000,
settings, data, 1,       0x2A0000, 0x010000,

I suppose that the module OTA installs the boot code in the ota_1 and the running code in the ota_0. 
What I need to change in this predefined setting ? and why the error "no update parition" ?