Moddable-OpenSource/moddable #158: [linker] Automatic stripping removes `Function`

jamieparkinson, 2019-03-29:
It seems like the automatic stripper in XSL removes too much of the `Function` object. When I compile the following with `"strip": "*"` -
```
const code = '42 * a';
const foo = new Function('a', code);

trace(foo(27));
```
I get an error as the `Function` constructor has been stripped (`Function: dead strip!`), when I expect to find `1134` traced.

Is this intended behaviour? If so it would be good if it was documented. Thanks!

phoddie, 2019-03-29:
@jamieparkinson - Thanks for the report. This just looks like an error in the rules `xsl` apples when stripping. We'll take a look.

winpuc, 2019-04-01:
Is this related to https://github.com/Moddable-OpenSource/moddable/issues/141... i.e. are Promises and Maps affected by the same problem?

phoddie, 2019-04-04:
The issue is corrected in our next push.

In the code above, `code` should be initialized to the following to give the expected result:

``````
  const code = 'return 42 * a;';
``````

Keep in mind that when automatic stopping is applied, unused engine features are eliminated. The build tools cannot predict what will be passed to `eval` to evaluate. If the script assigned to `code` is changed to invoke `Math.random()` that will fail in the above example with with a "dead strip!" error because `Math.random` is stripped.

jamieparkinson, 2019-04-05:
Oops, right you are re the evaluated string. Thank you!

closed 2019-04-05 by jamieparkinson