Moddable-OpenSource/moddable #160: Any plans to provide firmware files to flash?

wilberforce, 2019-04-11:
I tried to find bin flies to flash to the esp8266 or esp32 for testing - however it appears that you need to build from source?

Have you considered using travis to test changes and build firmware files?

phoddie, 2019-04-11:
Each application to run is directly linked to the firmware. Consequently, binary file provided would only run a single application. What do you hope to test by having a binary image provided?

wilberforce, 2019-04-11:
Ah - i understand now. I've been using Espruino - you flash the firmware onto the device and then can interactively connect and access the javascript interterpreter.

I'll re-phrase my question - is there any plans to provide downloads of the build environment so that it can be downloaded and installed without building from source?

Thanks

lll000111, 2019-04-12:
What is the purpose? You cannot skip the build step because *everything* including your JS app is build. You cannot not have a build step. Sure, you can safe the (pretty quick) step of compiling the build environment itself, but what for?

wilberforce, 2019-04-12:
> What is the purpose? 
There is not an easy set up - you need to follow a quite a few steps and even for someone familiar with the tool chains and compiling - It's quite a commitment to trial moddable. I was just wondering if there was an easier path.

> low.js
Thanks - I'm aware of that project. It requires the psram version of the eps32 so does not not work on smaller hardware.

closed 2019-04-12 by wilberforce

phoddie, 2019-04-12:
@wilberforce - thanks for your patience in learning a bit about the Moddable SDK.

Our build is designed to keep the firmware as small as practical. That requires build-time tools with the ability to include on the language features used as part of the firmware. This also helps minimize RAM use, as much of the bookkeeping oof JavaScript objets and modules can move from RAM too Flash.ROM. Keeping the firmware and RAM footprint as small as possible is critical for delivering embedded products. It is indeed less convenient than pre-building a single environment with all possible features and then running scripts on top of that. We have found the results to be more than worth it.

> It's quite a commitment to trial moddable.

FWIW - in our experience in running workshops and talking with people who have set-up the SDK using our [Getting Start guide](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md#moddable-sdk--getting-started), it is often about 15 minutes. The most common issues tend to around the USB drivers to communicate with the development board, which is unfortunately not solved with prebuilt firmware images.

You might find [our REPL](https://github.com/Moddable-OpenSource/moddable/tree/public/examples/js/repl) of interest. Like other examples, you still have to build it from source but once that's done you can explore interactively. The REPL is introduced in our [blog post on eval](http://blog.moddable.com/blog/eval/) which includes information on how to add additional modules to the REPL.

In the long term we do hope to have easier paths available for developers to "kick the tires."

wilberforce, 2019-04-13:
Thanks - the intial reception was a bit frosty! I have persisted and have managed to push the hello world example to an ESP32 from windows 10 .
> The most common issues tend to around the USB drivers
I was 1 of 2 people who ported the ESP32 build ( using ESP8266 as a template ) to espruino so I'm familiar with tools and drivers – it’s just the learning curve into a new environment that has challengers!

Thanks for the links – I’ll take a look.

The next step is to see if I can get a web server serving files from the spiff file system, and it looks like there is currently no support for OneWire- I’m interested in interacting the temperature sensors.

phoddie, 2019-04-13:
>  I have persisted and have managed to push the hello world example to an ESP32 from windows 10 .

Excellent. Congrats.

> The next step is to see if I can get a web server serving files from the spiff file system 

I suggest starting with the http server example that [streams the response in chunks](https://github.com/Moddable-OpenSource/moddable/blob/public/examples/network/http/httpserverchunked/main.js) and replace the random data the example returns with file reads.