Moddable-OpenSource/moddable #161: Server Message Constants

wilberforce, 2019-04-15:
https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/network/network.md#callbackmessage-val1-val2-1

The message is an integer - Is there a reason that the class does not define the constants?

Looking at the Pins examples, there are constants within the class for various states.

Something like: 

```
Server.MsgDisconnected = -1;
Server.MsgConnected = 1;
Server.MsgStatus = 2;
Server.MsgHeader = 3;
Server.MsgHeaders = 4;
Server.MsgResponse = 8;
Server.MsgResFragment = 9;
Server.MsgComplete = 10;
```

I would think that this would  make the  code more readable - as the numbers on their own mean nothing.

Do you want a pull request for this sort of thing (after discussion?)

phoddie, 2019-04-16:
> The message is an integer - Is there a reason that the class does not define the constants?

There is a reason: efficiency. But, the benefit is small compared to the readability cost.

I defined constants for the client and server some time ago. I'll merge those in. Here they are for reference.

```
Server.connection = 1;
Server.status = 2;
Server.header = 3;
Server.headersComplete = 4;
Server.requestFragment = 5;
Server.requestComplete = 6;
Server.prepareResponse = 8;
Server.responseFragment = 9;
Server.responseComplete = 10;
Server.error = -1;

Request.requestFragment = 0;
Request.status = 1;
Request.header = 2;
Request.headersComplete = 3;
Request.responseFragment = 4;
Request.responseComplete = 5;
Request.error = -2;
```

N.B. The same work should be done for the other network protocol implementations. I've got notes for most of those too...

wilberforce, 2019-04-16:
Thanks.

If instead of using server.xxx  - they were defined as `const` would the compile then flatten to integers?

I'm surpised that `Server.headersComplete` is not flattened to 4 - however I'm not aware of the internals of xs!

phoddie, 2019-04-16:
While reducing `Server.headersComplete` to 4 in some circumstances, is not trivial.

N.B. There is no way to use `const` when declaring an integer property. `Object.freeze`to achieves that result (and we do that during [preload](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/preload.md)).

closed 2019-04-18 by wilberforce