Moddable-OpenSource/moddable #162: File examples and Spiffs on ESP32

wilberforce, 2019-04-15:
The files examples just use a filename 'test.txt', however I have discovered that a path needs to be supplied for both the the `win` and `esp32`  cases.

For the esp32 - it needs to be `/spiffs/test.txt` and on the desktop you can specify `./folder/test.txt` and it will read/write in the the current project folder.

`./` also works for windows - `c:\\users\....\\` is not necesssary.

For the ESP32 you also need to add the partitions csv to the manifest. My current manifest looks like this:

```
{
	"platforms":{
		"esp32": {
			"build": {
				"PARTITIONS_FILE": "$(MODDABLE)/examples/files/files/partitions.csv",
			},
			"config": {
				"root_fs": "/spiffs/",
			},
		},
		"win": {
			"config": {
				"root_fs": "./dist/",
			},
		},		
	},
...
}
```

so this alows:

```
import config from "mc/config";
const root = config.root_fs;
```

Which keeps the code platform independant. Will this sort of thing confuse the basic examples or should it be constributed as  separate example? The current examples hint at this with commented out code that could be fleshed out.

bfriedkin, 2019-04-16:
Hello -

Thank you for the good suggestion. Defining the root directory in the "config" definitely makes sense and cleans-up the example source.

We ended up putting the root directory in the Files module's manifest.json, so that the directory is always available to apps. An application can always override the directory by including a "config" section and associated properties in the application manifest.json file. We've attached our modified Files module's manifest.json [here](https://github.com/Moddable-OpenSource/moddable/files/3086443/manifest.json.zip) for reference. The change will be available in our next OSS push. We also slightly modified the config "root" property layout to allow additional file system properties:

```
import config from "mc/config";

const root = config.fs.root;
```

Regards,
Brian

wilberforce, 2019-04-17:
Excellent - I'll update my config to mirror.

How often do you push to this repository? I've notice that quite a few of the pushes from other's contributions have been done with a force push so you can't tell what the contributions file contains as the history is scrambled.

phoddie, 2019-04-17:
> How often do you push to this repository?

It really depends. Every week or two, looking backwards.

>  I've notice that quite a few of the pushes from other's contributions have been done with a force push so you can't tell what the contributions file contains as the history is scrambled.

That was resolved a few weeks back. It should not be an issue moving forward,.

closed 2019-04-17 by wilberforce