Moddable-OpenSource/moddable #163: ESP32 Spiffs Filesystem Size

wilberforce, 2019-04-15:
The example `partitions.csv` for the ESP32 has spiffs with a size of 60K.

I would like to increase this considerably ( At least 256K ).

As this is defined as the last partition there is not the space to simply increase,

Name, | Type, | SubType, | Offset, | Size H, | Size D, | Next
-- | -- | -- | -- | -- | -- | --
nvs, | data, | nvs, | 0x9000, | 0x006000, | 24K, | 0xF000
phy_init, | data, | phy, | 0xf000, | 0x001000, | 4K, | 0x10000
factory, | app, | factory, | 0x10000, | 0x3B0000, | 3776K, | 0x3C0000
xs, | 0x40, | 1, | 0x3D0000, | 0x010000, | 64K, | 0x3E0000
settings, | data, | 1, | 0x3E0000, | 0x010000, | 64K, | 0x3F0000
storage, | data, | spiffs, | 0x3F0000, | 0x0F000, | 60K, | 0x3FF000

Without knowing future plans - is there any reason the factory app is so large?

If is reduced in the partition table there must be somewhere else that needs to be updated to specify the available code space or is this ready from the partition table?

Can the spiffs be moved after factory without side effects?

Thanks!

phoddie, 2019-04-15:
I think you are referring to [`partitions.csv`](https://github.com/Moddable-OpenSource/moddable/blob/public/build/devices/esp32/xsProj/partitions.csv) in the File example, not the SDK default [`partitions.csv`](https://github.com/Moddable-OpenSource/moddable/blob/public/build/devices/esp32/xsProj/partitions.csv) (which has no SPIFFS partition at all).

The partitions are configurable because it is nearly impossible to get two people to agree on the what they should be. ;)

> Without knowing future plans - is there any reason the factory app is so large?

In general, we make the factory partition as big as we can because it is where the developer's application lives. That's usually the biggest thing. If you are indeed referring to the File example, it makes sense to increase the SPIFFS partition there to give developers more space to explore with that example.

> Can the spiffs be moved after factory without side effects?

Rearrange and resize the partitions as you like for your project. The Moddable SDK adds no additional constraints to the partitions beyond those imposed by the ESP32 IDF.

wilberforce, 2019-04-16:
Thanks for answering - yes I was talking about the spiffs example. 

I'll make the factory app smaller then. I though that there might be somewhere else to update to tell the build system what the maximum size of the firmware can be.

phoddie, 2019-04-16:
As far as we've seen, updating the partitions.csv is enough. The IDF build system manages to propagate that all the way through.

closed 2019-04-20 by phoddie