Moddable-OpenSource/moddable #164: Win 10 VM crashing

wilberforce, 2019-04-17:
I'm looking for help to start debugging this. The `mc` in xsbug disappears, so I believe the background task that is running the VM has crashed.
Is there a way to get a log output - or a verbose output or something? I'm unsure on were to start.

Thanks.

phoddie, 2019-04-17:
If it is a hard crash, you should dump out to the (very minimal) gdb from the ESP32 IDF. It is often good enough too get a stack trace. Do you see that?

wilberforce, 2019-04-17:
Thanks - Sorry - the crash is when the target is `-p win`.

the `mc` detaches from the debugger

phoddie, 2019-04-17:
> the mc detaches from the debugger

I assume you are referring to xsbug, not the native debugger.

I don't debug on Windows these days. @bfriedkin can help there. On macOS, FWIW, I first launch the simulator from the Xcode IDE with the debugger, then run `mcconfig` to build and launch the JavaScript project. If there's a crash, it lands in the Xcode debugger.

wilberforce, 2019-04-17:
mcconfig starts - the `mc` tab appears in xsbug  and output appears in the log. However after certain http requests, the log stops outputting and the `mc` tab disappears. If I start `mcconfig -d -m -p win` it makes a new instance and re-attaches.

phoddie, 2019-04-17:
Right. That it is detaching from xsbug usually means that the socket connected between the target and xsbug was closed, which happens when there is a crash in the native code. The easiest way to see what is happening is to connect a native debugger to the target process.

wilberforce, 2019-04-17:
Ok. I need to find out how to that from visual studio code. 
Do you launch xsbug or mcconfig from the native debugger

phoddie, 2019-04-17:
> Do you launch xsbug or mcconfig from the native debugger

Neither. Launch the simulator executable (I think that's at $MODDABLE/build/bin/win/debug/simulator.exe) with the native debugger. Then use `mcconfig` to have the simulator run the JavaScript project.

bfriedkin, 2019-04-17:
Hello -

On Windows, you first launch Visual Studio and select File > Open > Project/Solution... and select the debug version of the Moddable SDK simulator. You can find that at %MODDABLE%\build\bin\win\debug\simulator.exe. Press the Start button to launch the simulator. At this point you can open C files and set breakpoints if you'd like. Breakpoints may display as an "open" circle, but get resolved when the Moddable app launches.

Then from a "Developer Command Prompt for VS 2017" console, build and launch the app, for example:

```
cd %MODDABLE%\examples\network\http\http\httpget
mcconfig -d -m -p win
```

When the app crashes, you'll hopefully get a stack trace in Visual Studio.

Regards,
Brian

wilberforce, 2019-04-17:
@bfriedkin - thanks very much.
I know get a stack trace of the simulator crashing.

The order of events:

a. start the `mcconfig -d -m -p win`
b. request http://localhost/
c. All ok 
d. request http:/.localhost/ again.
e. The VM crashes here in \modules\network\http\http.js:109

```
	close() {
		if (this.socket)
			this.socket.close();
		delete this.socket;
		delete this.buffers;
		delete this.callback;
		this.state = 11;  /// <<< crash
	}
};
f. Now the simulator is not in a good state
g. restart  `mcconfig -d -m -p win`
h. crash in simulator:

Exception thrown at 0x0F94C430 in simulator.exe: 0xC0000005: Access violation executing location 0x0F94C430. occurred

```
txSlot* instance = (flag) ? the->stack->value.reference : fxToInstance(the, the->stack);
	txSlot* property = mxBehaviorGetProperty(the, instance, (txID)id, index, XS_ANY); // <<< crash
	if (!property) {
		the->stack->kind = XS_UNDEFINED_KIND;
	}
```

So is this pattern correct:

```
let server = new Server({});
server.callback = function (message, value, etc) {
...

```

There appears to be an issue for the 2nd request with the object state?

bfriedkin, 2019-04-18:
@wilberforce  -

Thank you for the additional details. I have been able to reproduce a similar problem running the [httpserver](https://github.com/Moddable-OpenSource/moddable/tree/public/examples/network/http/httpserver) example app on Windows.

It appears that our Windows socket implementation might be inadvertantly terminating the underlying native WinSock network package when there are multiple sockets being opened and closed.

I would appreciate if you'd be able to try a workaround. In the file %MODDABLE%\modules\network\socket\win\modSocket.c, please comment out the call to WSACleanup here:

```
closesocket(xss->skt);
xss->skt = INVALID_SOCKET;
//WSACleanup();
```

Please let me know if that seems to resolve the problem. In the meantime I will work towards a proper fix.

Thanks,
Brian

wilberforce, 2019-04-18:
@bfriedkin 

Brilliant - yes that works!
Something is up - as I have -21 sockets in the instrumenation!

Each request the socket count decreases by one.

By the way - I have managed to get a build compiling under the Bash shell for windows 10. the tools all build my mcconfig reports unsupported platform.

```
root@office-pc:~/repos/www-server# mcconfig -d -m -p lin
# error: manifest_net - unsupported platform!
### Error: incompatible platform!
```

bfriedkin, 2019-04-18:
@wilberforce -

We have resolved both the WinSock and Instrumentation socket count issues. Fixes will be available in the next OSS push. Thank you for reporting the issues and for your patience.

Because each target platform is unique, not all examples/features work on all targets. The Linux target does not support the socket/net modules. Hence the errors reported by mcconfig.

Regards,
Brian

phoddie, 2019-04-22:
Fix pushed.

closed 2019-04-22 by phoddie