Moddable-OpenSource/moddable #166: [Question] Is Wi-Fi promiscuous mode supported on esp32/esp8266?

horihiro, 2019-04-20:
Do you have an example if it is supported?

phoddie, 2019-04-20:
There isn't support for promiscuous mode in the SDK. In theory, that should be straightforward to implement. I tried implementing it once for ESP8266, but it was never stable enough to be interesting. Even with a do-nothing callback to `wifi_set_promiscuous_rx_cb` the device crashes in seconds to minutes with what appears to be heap corruption.

```
# EXCEPTION DESCRIPTION
# Exception 3 LoadStoreError: Processor internal physical address or data error during load or store

# EXCEPTION LOCATION
# 0x40201aa9: umm_assimilate_up at /Users/hoddie/esp/esp8266-2.3.0/cores/esp8266/umm_malloc/umm_malloc.c:1163

# CALLS
# 0x40201b88: _umm_free at /Users/hoddie/esp/esp8266-2.3.0/cores/esp8266/umm_malloc/umm_malloc.c:1287
# 0x40201fec: free at /Users/hoddie/esp/esp8266-2.3.0/cores/esp8266/umm_malloc/umm_malloc.c:1733
# 0x40106d70: vPortFree at /Users/hoddie/esp/esp8266-2.3.0/cores/esp8266/heap.c:18
# 0x4022ac97: loop_task(ETSEventTag*) at /Users/hoddie/esp/esp8266-2.3.0/cores/esp8266/core_esp8266_main.cpp:110
```

 That discouraged me from trying on ESP32.

horihiro, 2019-04-21:
Thanks!
I understood that.

closed 2019-04-21 by horihiro