Moddable-OpenSource/moddable #171: Cross platform - how to handle `xsUnknownError` for unimplemented features

wilberforce, 2019-04-26:
On the ESP32 - this is fine, trying to dump available space in the filesystem.
```
try {
let info = System.info();
trace(JSON.stringify(info),'\n\n');
} catch {
	trace('no system info');
}
```

however it win it thows:
`..\moddable\modules\files\file\win\modFile.c (297) # Break: (host): unimplemented!`

even though it's in a try/catch

What is the best way to handle cross platform differences when developing?

phoddie, 2019-04-26:
> however it win it thows:...even though it's in a try/catch

A try/catch block catches exceptions. It does not suppress them. You can configure xsbug to [not break on exceptions](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/xsbug.md#preferences-pane) in the Preferences Pane.

> What is the best way to handle cross platform differences when developing?

There's no single answer to that. The Moddable SDK doesn't try to provide a completely uniform porting layer. That's perhaps impossible for embedded systems, and would make the runtime bigger and slower.

In this case, one answer is to implement the File System `info` call. Windows certainly provides that functionality.

Another approach is to define properties in the `config` section of the manifest for various features. Each target platform can have its own `config` in the manifest. This is how [config.file.root](https://github.com/Moddable-OpenSource/moddable/blob/public/modules/files/file/manifest.json) is defined to be different on each platform.

```
   import config from "mc/config";

   let info;
   if (config.hasFileSystemInfo)
     info = System.info();
```

Another approach is to have a different version of [file.js](https://github.com/Moddable-OpenSource/moddable/tree/public/modules/files/file) for each target platform.  Then you could simply omit the `info` function entirely from the prototype for targets that do not implement it. That way you can check for the function directly:

```
   let info;
   if (System.info)
     info = System.info();
```
 
That approach is probably the most natural in JavaScript. It is also smaller -- any stub functions are eliminated. It will be some ongoing work to ensure the prototype for File doesn't diverge across platforms beyond omitting unimplemented functions. Still, it could be worth trying out.

wilberforce, 2019-04-26:
Thanks - that makes a lot of sense.

I want to be able to `fake` hardware or a ble device on windows.

Is there anything predefined like node's `process.env` . I've just added to my manifest:

```
{
	"platforms": {
		"esp": {
			"config": {
				"platform": {
					"name": "esp8266"
				}
			},
		},
```

Which strikes me might already exist?

phoddie, 2019-04-26:
We've intentionally avoided having a platform identifier in the configuration. The reason is because testing the host platform name isn't a reliable way to test for features. Classic Mac OS added [`Gestalt`](https://en.wikipedia.org/wiki/Gestalt_(Mac_OS)) in System 6 to address this same issue. In the case of your Windows file system example, checking for the Windows platform would be incorrect once the `info` was implemented.

You are free to add a platform identifier to your project manifests, of course. ;)

closed 2019-04-26 by wilberforce