Moddable-OpenSource/moddable #172: [Need help] How to get large size content with HTTPS request

horihiro, 2019-04-27:
I'm porting [google-tts](https://github.com/zlargon/google-tts) to moddable platfrom.

The original library gets a key from [Google Translate](https://translate.google.com/).
So I want to do that on moddable SDK.

But `!!! xs: failed to allocate 1504 bytes for chunk !!!` exception occurs when esp32 with moddable SDK tries to get the Google Translate page, which content size is 54Kbytes.

It seems that esp32 cannot allocate memory for such a large content.
\# The source is same as [httpsget example](https://github.com/Moddable-OpenSource/moddable/blob/public/examples/network/http/httpsget/main.js), except URL.

Could you please let me know how to get the large content using http/https if you know?

Best Regards,

wilberforce, 2019-04-27:
use this example instead....  you need to read the response in chunks....

https://github.com/Moddable-OpenSource/moddable/blob/public/examples/network/http/httpsgetstreaming/main.js

horihiro, 2019-04-27:
Hi @wilberforce !
Thank you for your quick response!!

I will try to handler the response as a stream.

> Why is it 54K - what does it contain?

Sorry, I can't answer to your question because [the page](https://translate.google.com/) was created by Google.
I found the size by `Content-Length` header.
[image]

The page contains HTML/CSS/JS.

closed 2019-04-27 by horihiro

wilberforce, 2019-04-27:
There is probaby a json API you can call - rather than using the html version of the page

phoddie, 2019-04-27:
The default XS allocation on ESP32 is modest, just [32 KB](https://github.com/Moddable-OpenSource/moddable/blob/public/examples/manifest_base.json#L9). That is the same as ESP8266. It works for all the examples and encourages developers to be aware of the memory use of the project. You can increase that in your manifest.

The ESP32 has plenty of memory. The `httpsget` example has about 200 KB free system memory when running. On ESP32, because of the way internal RAM is banked, the maximum contiguous block appears to be about 130 KB. Add these lines to your manifest to use those:

	"creation": {
		"static": 134000,
	}

That is likely enough to load your 54 KB web page as a string. Note that the way the HTTP client is implemented there are two copies of the page in RAM temporarily -- each individual buffer as it arrives, and then a single string containing all of them concatenated together. At the application level you could avoid the extra copy with a little more work. I'm happy to provide an example of that, if it would help.