Moddable-OpenSource/moddable #174: Ball not rendered correctly in m5stack_imu example

zBuffer, 2019-05-02:
The ball is [blue](https://pbs.twimg.com/media/D5ap04zUUAA813r.jpg).

I notice if you set the rotation to 90 degrees the code will not run - because rotation = 0 will be passed to the Poco constructor in the demo code (/examples/drivers/m5stack-imu/main.js) and it will throw an exception telling the rotation is not supported.

But if you change the first lines of code to:
`import parseBMF from "commodetto/parseBMF";
import parseBMP from "commodetto/parseBMP";
import Poco from "commodetto/Poco";
import Resource from "Resource";
import ILI9341 from "ili9341";
import config from "mc/config";

let pixelsOut = new ILI9341({});
const GYRO_SCALER = 0.002;
const MAG_SCALER = 0.002;

let render = new Poco(pixelsOut, { rotation: config.rotation });
const width = render.width;
const height = render.height;`

The ball will now render correctly. Which is VERY odd.

I'm testing this on a M5Stack Gray device, which should have almost the same hardware (just without NeoPixels and PSRAM) as the new M5Stack Fire. I understand the existing code is based on the old M5Stack Fire, which I have the IMU driver code slightly tweaked for the demo to work - but these are quite unrelated right?

zBuffer, 2019-05-02:
```c
void ili9341Send_16LE(PocoPixel *pixels, int byteLength, void *refcon)
{
	spiDisplay sd = refcon;
	modSPISetSync(&sd->spiConfig, byteLength > 0);
	//modSPITxSwap16(&sd->spiConfig, (void *)pixels, (byteLength < 0) ? -byteLength : byteLength);
	modSPITx(&sd->spiConfig, (void *)pixels, (byteLength < 0) ? -byteLength : byteLength);
}
```
I think it's somehow related to modSPITx supposed to expect little-endian data. Changing this portion of code fixes the issue without (`-r 90`).

andycarle, 2019-05-02:
This is curious, indeed. 

My best guess is that the M5Stack Gray has a slightly different display controller than the M5Stack Fire that we've been testing against. Consequently, it probably needs to be configured slightly differently. 

Here's something to try. In the [m5stack_fire's manifest](https://github.com/Moddable-OpenSource/moddable/blob/public/build/devices/esp32/targets/m5stack_fire/manifest.json), there is a section that specifies a configuration for the IlI9341 (`defines/ili9341`). That was recently updated to improve the display on the M5Stack Fire, but those changes may not have been correct for the M5Stack Gray.

Could you try replacing the `ili9341` section in that file with the one from [this version](https://github.com/Moddable-OpenSource/moddable/blob/d2bd27b9d695e7965c780e2881012bb022658a66/build/devices/esp32/targets/m5stack_fire/manifest.json)? If that fixes the issue we could make a separate target for the M5Stack Gray that uses that configuration.

zBuffer, 2019-05-02:
Just tried that version, produces display artifacts (similar to one of the issues that gave birth to that update), but the ball is still blue.

zBuffer, 2019-05-02:
I wanted to try passing `-f rgb565be` but png2bmp would't work, the example will not build.

zBuffer, 2019-05-02:
Changing `ili9341Send_16LE` will break the `examples/piu/cards` display.

andycarle, 2019-05-02:
This is all very interesting.

I'm sticking with my guess that this is a display controller configuration issue. But it's hard for me to sort out what the problem might be without the relevant hardware to test against. 

I'm looking through the [m5stack sources on github](https://github.com/m5stack/M5Stack/tree/master/src) and I see that they do, indeed, include support for several different display controllers. One thing that caught my eye: some of the recent devices use [this eSPI TFT graphics library](https://github.com/m5stack/M5Stack/blob/master/src/utility/In_eSPI.cpp) in addition to the ILI9341 driver. That library uses the ILI9341's display inversion command in some situations.

You might try adding one of these to the (new) `registers` section of the m5stack_fire manifest:
`"0x21, 0,",` or
`"0x20, 0,",`

So, the resulting block might look like:
```
"registers": [
	"0x01, 1, 0xA8,",
	"0xFF, 5,",
	"0xEF, 3, 0x03, 0x80, 0x02,",
	"0xCF, 3, 0x00, 0XC1, 0X30,",
	"0xED, 4, 0x64, 0x03, 0X12, 0X81,",
	"0xE8, 3, 0x85, 0x00, 0x78,",
	"0xCB, 5, 0x39, 0x2C, 0x00, 0x34, 0x02,",
	"0xF7, 1, 0x20,",
	"0xEA, 2, 0x00, 0x00,",
	"0xC0, 1, 0x23,",
	"0xC1, 1, 0x10,",
	"0xC5, 2, 0x3e, 0x28,",
	"0xC7, 1, 0x86,",
	"0x36, 1, 0xA8,",
	"0x3A, 1, 0x55,",
	"0xB1, 2, 0x00, 0x13,",
	"0xB6, 3, 0x08, 0x82, 0x27,",
	"0xF2, 1, 0x00,",
	"0x26, 1, 0x01,",
	"0xE0, 15, 0x0F, 0x31, 0x2B, 0x0C, 0x0E, 0x08, 0x4E, 0xF1, 0x37, 0x07, 0x10, 0x03, 0x0E, 0x09, 0x00,",
	"0xE1, 15, 0x00, 0x0E, 0x14, 0x03, 0x11, 0x07, 0x31, 0xC1, 0x48, 0x08, 0x0F, 0x0C, 0x31, 0x36, 0x0F,",
	"0x11, 0,",
	"0x21, 0,",
	"0xFF, 120,",
	"0x29, 0,",
	"0xFF, 0,"
],
```
Try both 0x21 and 0x20 to see what impact that command has.

zBuffer, 2019-05-02:
Changing the 0x21 register messes up the colors of the entire display. Changing 0x20 does nothing. I somehow doubt it's the display controller because the `examples/piu/cards` demo display fine.

The fact that `poco.drawMasked()` behaves differently for 0, 180, 270 degrees rotation than 90 degree rotation doesn't raise any concern at all? The rotation configuration doesn't even reach driver level.

zBuffer, 2019-05-02:
Surprise surprise.. `poco.drawBitmap()` is drawing the colors correctly.

zBuffer, 2019-05-03:
I've dug through the code and really couldn't find what went wrong. Is there a quick way for me to dump out the pixels sent to the driver? That way we can debug this together.

phoddie, 2019-05-03:
> Is there a quick way for me to dump out the pixels sent to the driver 

```
void ili9341Send_16LE(PocoPixel *pixels, int byteLength, void *refcon)
{
	int i, count = ((byteLength < 0) ? -byteLength : byteLength) >> 1;
	for (i = 0; i < count; i++)
		modLogHex(pixels[i]);
// etc.
```

andycarle, 2019-05-03:
@zBuffer Could you give us some more information about your build environment? What platform are you building on? Is there anything unusual about your toolchain or anything?

Something else to try: I've uploaded a binary of the m5stack-imu example that I built [here](https://www.dropbox.com/s/6mrnqx5zpfex7j5/xs_esp32.bin?dl=0). If you could flash it on to your device (using a command like `esptool.py --chip esp32 --port /dev/cu.SLAB_USBtoUART --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x10000 ./xs_esp32.bin` ) and see what results you get, that would help us sort out if the issue is specific to the M5Stack Gray device or if it is something about your build.

zBuffer, 2019-05-04:
Probably the only thing that’s unusual is the fact I’m building it on a
32-bit Windows 7 VM. There are a few depreciation warnings during the build
but that’s about it.

I haven’t tried the binary you have built because I’m away from the device
right now. But as a quick check did you disable the IMU routines (or change
the whoami response) so it can run without throwing exceptions on M5Stack
Gray?

andycarle, 2019-05-04:
No, sorry, I did not. You may be able to get far enough to check the rendering by running without xsbug open or by continuing through the exceptions. I'm not in a position to do so right now, but I'll generate a new binary as soon as I can.

zBuffer, 2019-05-05:
> No, sorry, I did not. You may be able to get far enough to check the rendering by running without xsbug open or by continuing through the exceptions. I'm not in a position to do so right now, but I'll generate a new binary as soon as I can.

It's thrown in a constructor, a fatal exception if unhandled. I'm not sure how to force it to continue.

zBuffer, 2019-05-05:
Except of pixel dump with `poco.drawMasked()`
```
0
0
0
8e2
2652
4ea
4ea
25fb
25fb
5fb
5fb
5fb
5fb
c4f2
c4f2
c4f2
c4f2
44ea
44ea
84aa
84aa
64e2
64e2
24d2
24d2
e9b2
4886
0
0
0
0
0
```

Excerpt of pixel dump with `poco.drawBitmap()`
```
cb28
cb28
cb28
d348
fb26
eae4
9204
fb25
fb25
fb05
fb05
fb05
fb05
f2e5
f2c4
f2c4
eac4
eac4
ca44
aa04
ea84
e263
da64
d244
ca24
b223
9ae9
bc6f
bc4f
bc4f
ffbd
ffff
```

zBuffer, 2019-05-05:
Another sample:
`drawMasked`
```
7df7
e4ea
e4ea
e4f2
e4f2
e5f2
e5f2
e4f2
e4f2
a5fb
a5fb
c5a1
c5a1
63ea
63ea
64e2
64e2
e3da
e3da
2399
2399
e3ca
e3ca
c2b9
c2b9
62a9
62a9
a291
a291
d5af
a46c
0
```

`drawBitmap`
```
c2a7
eac4
fae4
f2e4
f2e4
f2e5
f2e5
f2e5
f2e4
fb05
caa5
a1e4
f2c5
ea84
e263
e284
e264
da43
b9e3
99c4
da23
ca03
c1e3
b9e3
b1c2
a9a2
6962
91a3
99a2
91e4
63f6
ffff
```

phoddie, 2019-05-05:
The change to line 2318 in commodettoPocoBlit.c should not make a difference. In fact, it should get you the wrong pixel 50% of the time.  If you look at the code above, the `srcPixel` you replaced with `*src` comes is just a long aligned read from `*src`.

Perhaps, somehow, the resource data is not long aligned. Trace the value of `src` to see.

```
modLogHex(src);
```

Also, you can check the resource itself in `mc.resources.c` in the build results. For the balls example, you should see

```
	{"balls-color.bmp", _0, sizeof(_0)},
```

That tells you that the resource data is in the variable `_0`. Looking at `_0` you should see it forced to 4 byte alignment:

```
static const unsigned char _0[] __attribute__((aligned(4))) = {
	0x42, 0x4d, 0x66, 0x1c, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x46, 0x00, 0x00, 0x00, 0x38, 0x00,
```

You can also check the link map output in `xs_esp32.map`. That should be a long aligned address:

```
 .rodata._0     0x000000003f41fcb4     0x1c66 ..../debug/idf/xs_esp32.a(mc.resources.c.o)
```

zBuffer, 2019-05-05:
> The change to line 2318 in commodettoPocoBlit.c should not make a difference. In fact, it should get you the wrong pixel 50% of the time. If you look at the code above, the `srcPixel` you replaced [...]

Now that you mentioned this, this warning did pop up during build time:
```
\moddable\examples\drivers\m5stack-imu>mcconfig -d -m -p esp32/m5stack_gray
        1 file(s) copied.
# xsc ili9341.xsb
# xsc main.xsb
# xsid modIli9341.c.xsi
# xsl modules
# cc mc.xs.o (slots in flash)
\Local\Temp\ccUX6yYX.s: Assembler messages:
\Local\Temp\ccUX6yYX.s:17556: Warning: setting incorrect section attributes for .rodata.mod.1
# cc modIli9341.c.o
# ld xs_esp32.bin
        1 file(s) copied.
```

phoddie, 2019-05-06:
The warning isn't a problem. The lack of an `aligned` attribute is. I believe that's due to this code in the [mcrez](https://github.com/Moddable-OpenSource/moddable/blob/public/tools/mcrez.js#L137) tool:

```
	if (this.windows)
		file.line("static const unsigned char _", index, "[] = {");
	else
		file.line("static const unsigned char _", index, "[] __attribute__((aligned(4))) = {");
```

The intent of that is to exclude the `aligned` attribute when build for Windows because the Microsoft compiler doesn't support that GCC feature. If we are lucky, the ESP32 problem should go away if you replace the above with:

```
	file.line("static const unsigned char _", index, "[] __attribute__((aligned(4))) = {");
```

That's not a good fix, though. It should be something like

```
	if (("win" === this.platform) || ("x-win" === this.platform)))
		file.line("static const unsigned char _", index, "[] = {");
	else
		file.line("static const unsigned char _", index, "[] __attribute__((aligned(4))) = {");
```

Fingers crossed.

zBuffer, 2019-05-06:
Changing line 137 to:
```
				if (("win" === this.platform) || ("x-win" === this.platform))
					file.line("static const unsigned char _", index, "[] = {");
```
fixes the issue. Mystery solved!

phoddie, 2019-05-06:
Thanks for your persistence and excellent sleuthing to track this down. We'll integrate the fix.

closed 2019-05-06 by zBuffer