Moddable-OpenSource/moddable #176: Anyone using Sonoff?

wilberforce, 2019-05-02:
I've just got my hands on a 1 ch sonoff and got it working usng the factory firmware.
Next challenge is to get moddable onto it .
I think it uses the esp8266 variant and 1mb of flash. 
Before I solder headers and and FTDI-> usb thingy, (which It think will be a bit of phff to flash as it will need the glash button to be hit at power up) I'm wondering if anyone has got ota esp8266 update moddable working - I remember seeing esp32 code - not sure about the esp826?

 With only 1mb I guess the map would have to be a 512/512 for ota , size size might be a bit of a squeeze and I might be dreaming. Sorry this is a bit of a rant  - however there isn't a forum to discuss such things!

Oh - and can C++ libraries be used with moddable? I'm looking to port a Dallas Onewire to the esp8266 - however the auduino one is C++.

jim80, 2019-05-04:
" I'm looking to port a Dallas Onewire to the esp8266 "
Check the current pull request for this.

wilberforce, 2019-05-04:
@jim80

You mean the pull request I opened :-;

https://github.com/Moddable-OpenSource/moddable/pull/170

That's for the esp32 - I'm working on the esp8266 port now

jim80, 2019-05-05:
Ha! Ooops. Please forgive my lack of attention!
I'll be pleased to see the 8266 port, many thanks for that.

wilberforce, 2019-05-05:
Using this manifest fragment, the sonoff is working!

```
	"esp/8285": {
	        "build": {
                "FLASH_SIZE": "1M",
                "FLASH_LAYOUT": "eagle.flash.1m.ld"
				},
			"config": {
				"led1_pin": 13,
				"button1_pin": 0,
			},
			"modules": {
				"*": "$(MODULES)/pins/digital/monitor/esp/*",
			},
		},	
```

[https://trello-attachments.s3.amazonaws.com/55b2a8dc6d979de5a73970a3/5cce9cd8a651197d0688158b/1f9bfb2368709e15041bb7b7b5a1c4ea/IMG_5970.JPG](url)

phoddie, 2019-05-05:
Cool. Congrats. Apologies for not replying sooner. That linker script is in the project from work we did on Sonoff products last year. We haven't documented it yet, but it does work. ;)

I gave a couple talks about hacking the Sonoff bulb earlier this year, at an IoT event at MIT and a Scripts and Things meetup. I'll be [presenting it at Maker Faire](https://makerfaire.com/maker/entry/69518/) in a couple weeks.

closed 2019-05-05 by phoddie