Moddable-OpenSource/moddable #179: Cannot connecto to Moddable Zero

nicholasareed, 2019-05-04:
I received my Moddable Zero (ESP8266), but after re-trying the Mac install steps a few times, I keep getting stuck at building for the connected device. The device is running the Drag app by default (shows up as soon as it is plugged in). 

```
$ mcconfig -d -m -p esp/moddable_zero
# cpp core_esp8266_main.cpp
# cpp abi.cpp
# cc core_esp8266_phy.c
# cc core_esp8266_noniso.c
# cc cont_util.c
# cc cont.S
# cc core_esp8266_postmortem.c
# cc core_esp8266_si2c.c
# cc core_esp8266_timer.c
# cc core_esp8266_wiring.c
# cc core_esp8266_wiring_digital.c
# cc core_esp8266_wiring_pwm.c
# cpp Esp.cpp
# cc heap.c
# cc libc_replacements.c
# cc spiffs_cache.c
# cc spiffs_check.c
# cc spiffs_gc.c
# cc spiffs_hydrogen.c
# cc spiffs_nucleus.c
# cc time.c
# cc umm_malloc.c
# cpp Schedule.cpp
# cc bsearch.c
# cc e_fmod.c
# cc rtctime.c
# cc tinyprintf.c
# cc tinyuart.c
# ar libcirom.a
# cc xsHost.c (strings in flash + force-l32)
# cc xsPlatform.c (strings in flash + force-l32)
x - lib_a-setjmp.o
# cc xsAll.c (strings in flash + force-l32)
# cc xsAPI.c (strings in flash + force-l32)
# cc xsArguments.c (strings in flash + force-l32)
# cc xsArray.c (strings in flash + force-l32)
# cc xsAtomics.c (strings in flash + force-l32)
# cc xsBigInt.c (strings in flash + force-l32)
# cc xsBoolean.c (strings in flash + force-l32)
# cc xsCode.c (strings in flash + force-l32)
# cc xsCommon.c (strings in flash + force-l32)
# cc xsDataView.c (strings in flash + force-l32)
# cc xsDate.c (strings in flash + force-l32)
# cc xsDebug.c (strings in flash + force-l32)
# cc xsError.c (strings in flash + force-l32)
# cc xsFunction.c (strings in flash + force-l32)
# cc xsGenerator.c (strings in flash + force-l32)
# cc xsGlobal.c (strings in flash + force-l32)
# cc xsJSON.c (strings in flash + force-l32)
# cc xsLexical.c (strings in flash + force-l32)
# cc xsMapSet.c (strings in flash + force-l32)
# cc xsMarshall.c (strings in flash + force-l32)
# cc xsMath.c (strings in flash + force-l32)
# cc xsMemory.c (strings in flash + force-l32)
# cc xsModule.c (strings in flash + force-l32)
# cc xsNumber.c (strings in flash + force-l32)
# cc xsObject.c (strings in flash + force-l32)
# cc xsPlatforms.c (strings in flash + force-l32)
# cc xsProfile.c (strings in flash + force-l32)
# cc xsPromise.c (strings in flash + force-l32)
# cc xsProperty.c (strings in flash + force-l32)
# cc xsProxy.c (strings in flash + force-l32)
# cc xsRegExp.c (strings in flash + force-l32)
# cc xsRun.c (strings in flash + force-l32)
# cc xsScope.c (strings in flash + force-l32)
# cc xsScript.c (strings in flash + force-l32)
# cc xsSourceMap.c (strings in flash + force-l32)
# cc xsString.c (strings in flash + force-l32)
# cc xsSymbol.c (strings in flash + force-l32)
# cc xsSyntaxical.c (strings in flash + force-l32)
# cc xsTree.c (strings in flash + force-l32)
# cc xsType.c (strings in flash + force-l32)
# cc xsdtoa.c (strings in flash + force-l32)
# cc xsre.c (strings in flash + force-l32)
# cc xsmc.c (strings in flash + force-l32)
# cpp main.cpp
# xsc pins/digital.xsb
# xsc ili9341.xsb
# xsc xpt2046.xsb
# xsc preference.xsb
# xsc time.xsb
# xsc timer.xsb
# xsc Resource.xsb
# xsc instrumentation.xsb
# xsc main.xsb
# xsc commodetto/Bitmap.xsb
# xsc mc/config.xsb
# xsid digital.c.xsi
# xsid modGPIO.c.xsi
# xsid modSPI.c.xsi
# xsid modIli9341.c.xsi
# xsid modXpt2046.c.xsi
# xsid modPreference.c.xsi
# xsid modTime.c.xsi
# xsid modTimer.c.xsi
# xsid timer.c.xsi
# xsid Resource.c.xsi
# xsid modInstrumentation.c.xsi
# xsid commodettoBitmap.c.xsi
# xsid modGPIO.h.xsi
# xsid modSPI.h.xsi
# xsid spi_register.h.xsi
# xsid modTimer.h.xsi
# xsid modInstrumentation.h.xsi
# xsid commodettoBitmap.h.xsi
# mcrez resources
# xsl modules
Total resource size: 0 bytes
# cc mc.xs.c (slots in flash)
# cc digital.c (strings in flash)
# cc modGPIO.c (strings in flash)
# cc modSPI.c (strings in flash)
# cc modIli9341.c (strings in flash)
# cc modXpt2046.c (strings in flash)
# cc modPreference.c (strings in flash)
# cc modTimer.c (strings in flash)
# cc modTime.c (strings in flash)
# cc timer.c.o
# cc Resource.c (strings in flash)
# cc commodettoBitmap.c (strings in flash)
# cc modInstrumentation.c (strings in flash)
# cc mc.resources.c (slots in flash)
# ld main.bin
esptool.py v2.4.0
# Versions
#  ESP:   esp8266-2.3.0
#  XS:    ee60dd72-dirty
# Memory usage
#  Ram:    31304 bytes
#  Flash: 440388 bytes

esptool.py v2.4.0
Traceback (most recent call last):
  File "/Users/nickreed/esp/ESP8266_RTOS_SDK/components/esptool_py/esptool/esptool.py", line 3022, in <module>
    _main()
  File "/Users/nickreed/esp/ESP8266_RTOS_SDK/components/esptool_py/esptool/esptool.py", line 3015, in _main
    main()
  File "/Users/nickreed/esp/ESP8266_RTOS_SDK/components/esptool_py/esptool/esptool.py", line 2717, in main
    esp = ESPLoader.detect_chip(args.port, initial_baud, args.before, args.trace)
  File "/Users/nickreed/esp/ESP8266_RTOS_SDK/components/esptool_py/esptool/esptool.py", line 241, in detect_chip
    detect_port = ESPLoader(port, baud, trace_enabled=trace_enabled)
  File "/Users/nickreed/esp/ESP8266_RTOS_SDK/components/esptool_py/esptool/esptool.py", line 212, in __init__
    self._port = serial.serial_for_url(port)
  File "/Users/nickreed/Library/Python/2.7/lib/python/site-packages/serial/__init__.py", line 88, in serial_for_url
    instance.open()
  File "/Users/nickreed/Library/Python/2.7/lib/python/site-packages/serial/serialposix.py", line 268, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port /dev/cu.SLAB_USBtoUART: [Errno 2] No such file or directory: '/dev/cu.SLAB_USBtoUART'
make: *** [debugmac] Error 1
```

I've tried running the following to see if the device shows up:
```
$ ls /dev/cu.*
/dev/cu.Bluetooth-Incoming-Port
``` 
I'd expect `/dev/cu.SLAB_USBtoUART` to show up, but it never does. I've installed (and tried uninstalling, reinstalling, etc) the Silicon Labs CP210x USB to UART VCP driver, recreating all the folders, etc. 

Am I missing something simple? Enable flashing the device somehow?

nicholasareed, 2019-05-04:
Different USB port finally fixed the issue (can see/build at least!). Previously I had a different Moddable Zero on the first USB port. Conflict comehow?

lprader, 2019-05-06:
More likely just a problem with the USB port. I use a Mac too and the USB ports can be finicky. Switching ports usually works, although I'll occasionally have to restart my laptop when both quit on me.

closed 2019-05-06 by phoddie