Moddable-OpenSource/moddable #185: Missing support for ArrayBuffer in xs_i2c_write

zBuffer, 2019-05-09:
Took me a while (and loads of headless chicken debugs) to notice SMB writeBlock doesn't support ArrayBuffer (but readBlock supports it!).

Here's a suggestion `modules/pins/i2c/i2c.c:162`:
```
		if(xsmcIsInstanceOf(xsArg(i), xsArrayBufferPrototype))
		{
			l = xsGetArrayBufferLength(xsArg(i));
			if ((len + l) > sizeof(buffer))
				xsUnknownError("40 byte write limit");
			c_memmove(&buffer[len], xsmcToArrayBuffer(xsArg(i)), l);
			len += l;
		}
		else
		{
			xsmcGet(xsVar(0), xsArg(i), xsID_length);
			l = xsmcToInteger(xsVar(0));
			if ((len + l) > sizeof(buffer))
				xsUnknownError("40 byte write limit");
			for (j = 0; j < l; j++) {
				xsmcGet(xsVar(0), xsArg(i), j);
				buffer[len++] = xsmcToInteger(xsVar(0));
			}
		}
```

phoddie, 2019-05-09:
Sure, we can add `ArrayBuffer` support to I2C. Thanks for the suggestion.

> Took me a while (and loads of headless chicken debugs) to notice SMB writeBlock doesn't support ArrayBuffer

Apologies for the trouble. You might have spared a chicken by checking the documentation. The [documentation for SMB](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/pins/pins.md#smbus) states that its implementation is extension of I2C:

> The `SMBus` class extends the `I2C` class with additional functions.

The [documentation for I2C](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/pins/pins.md#writevalue--stop) does not suggest `ArrayBuffer` is supported for `write`. The supported type are listed:

> The values may be numbers, which are transmitted as bytes, Arrays, TypedArrays, and strings (which are transmitted as UTF-8 encoded text bytes).

zBuffer, 2019-05-09:
Whoops, my bad. Was expecting ArrayBuffer to be deriving class of Array or
TypedArray, and was expecting i2c write to behave similar to i2c read,
where it writes into an ArrayBuffer.

Also realised I may have avoided this embarrassment by initialising a
`Uint8Array`, passing `Uint8Array.buffer` to i2c read, and `Uint8Array` to
i2c write... and `DataView(Uint8Array.buffer)` for structural read/writes
to the buffer.

phoddie, 2019-05-10:
No problem. For consistency I2C should support `ArrayBuffer`. That's the fundamental binary type in JavaScript. I've got that locally and will push in the coming days.

closed 2019-05-13 by zBuffer