Moddable-OpenSource/moddable #186: Websocket Server supports write?

nicholasareed, 2019-05-10:
[Documentation](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/network/network.md#writemessage-1) seems to show it is easy to have a WS server write a message to connected clients, but I can't get it to work in practice. I checked the [source](https://github.com/Moddable-OpenSource/moddable/blob/public/modules/network/websocket/websocket.js#L277) seems to [show](https://github.com/Moddable-OpenSource/moddable/blob/public/modules/network/socket/socket.js#L37) that only the Client supports write. 

Am I missing something obvious? 

```
import {Server} from "websocket"
let wsServer = new Server({});
wsServer.write("hello"); // # Error: "Break: ?: call write: no function!"
```

wilberforce, 2019-05-10:
@nicholasareed 

Hi - had the same issue until I wokred out you need to call from the scope of the callback.

This is kind of hacky, as it only works for one client. I really needs to be a list of clients, but I've not has time to look at this as I think the connections are already managed by the listener, so i would be best to use that rather than maintaning a separate list...

```
import {
	Server as Ws
} from "websocket"

let ws = new Ws({
	port: 8081
});

let the_ws=null;
ws.callback = function (message, value, etc) {
	trace(JSON.stringify({
		message: message,
		value: value,
		etc: etc
	}), '\n');
	switch (message) {
		case 1:
			trace("main.js: socket connect.\n");
			break;

		case 2:
			trace("main.js: websocket handshake success\n");
			trace('sending:', JSON.stringify(model));
			this.write(JSON.stringify(model));
			the_ws=this;
			break;

		case 3:
			trace(`main.js: websocket message received: ${value}\n`);
			var update = JSON.parse(value);
			for (var o in update) {
				model[o] = update[o];
			}
			this.write(JSON.stringify(model));
			break;

		case 4:
			trace("main.js: websocket close\n");
			the_ws=null;
			break;
		case 5:
			trace("main.js: websocket sub-protocol\n");
			return 'esp';
			break;
	}
}

....

function send_to client(stuff) {
	if ( the_ws )
		the_ws.write(JSON.stringify(stuff));
}

```

or do the `ws.write` from in the callback.

closed 2019-05-10 by nicholasareed

phoddie, 2019-05-10:
> ...until I worked out you need to call from the scope of the callback

You can call `write` any time after a connection has been established. The object you call `write` on must be a valid client connection

```
import {Server} from "websocket";
let wsServer = new Server({});
wsServer.write("hello"); // # Error: "Break: ?: call write: no function!
```

In this case `wsServer` is not a connection to a client, but the server itself. Each client instance first appears when the `Client.connect` (value of 1) message is received. Recall that a server may have multiple simultaneous connections; they cannot all be accessed as `wsServer`.

> **Note**: `this` in the callback is a connection instance, not the server itself.

wilberforce, 2019-05-10:
@phoddie 

I'm even more confused! 

> You can call write any time after a connection has been established. The object you call write on must be a valid client connection

`write` on which on object? I've used breakpoints and found the outer server does not have a `write` method - are you saying this changes?.

Perhaps an example that does a broadcast to more that on client would be useful.