Moddable-OpenSource/moddable #189: Possible missing nvs_flash_init() in esp32/modPreference.c

zBuffer, 2019-05-13:
`xs_preference_set()` does not take into account if NVS partition is not initialized (`ESP_ERR_NVS_NOT_INITIALIZED`).

What worked for me:
```
	err = nvs_open(xsmcToString(xsArg(0)), NVS_READWRITE, &handle);
	if(ESP_ERR_NVS_NOT_INITIALIZED == err) {
		err = nvs_flash_init();
		if (ESP_OK != err){
			if(err == ESP_ERR_NVS_NEW_VERSION_FOUND) {
				const esp_partition_t *part = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_NVS, "nvs");
				if(part) {
					err = esp_partition_erase_range(part, 0, part->size);
					if (ESP_OK != err)
						xsUnknownError("esp_partition_erase_range fail");
				}
				err = nvs_flash_init();
			}
			if (ESP_OK != err)
				xsUnknownError("nvs_flash_init fail");
		}
		err = nvs_open(xsmcToString(xsArg(0)), NVS_READWRITE, &handle);
	}
	if (ESP_OK != err)
		xsUnknownError("nvs_open fail");
```

Or is there an option to force esptool to re-flash nvs partition?

phoddie, 2019-05-13:
I'd like to better understand the problem you are reporting before considering solutions.

The NVS software component is always initialized. You can see `nvs_flash_init ` is called [very early](https://github.com/Moddable-OpenSource/moddable/blob/public/build/devices/esp32/xsProj/main/main.c#L207) in main.c.

The [documentation](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/storage/nvs_flash.html?highlight=nvs_flash_init#_CPPv414nvs_flash_initv) for `nvs_flash_init` indicates it initializes the default NVS partition if necessary. You can confirm this works by erasing the entire flash prior to running the preference sample:

```
esptool.py erase_flash
```

zBuffer, 2019-05-14:
Ah, if that's the case errors during `nvs_flash_init()`c (for my case, `ESP_ERR_NVS_NEW_VERSION_FOUND`c) weren't handled in `main.c`. Is there an argument available to be passed to `mcconfig` to run `erase_flash` prior to flashing the code?

phoddie, 2019-05-14:
This is a rollback situation - going from a newer version of IDF to a previous version. That seems unlikely in production. Also, it seems dangerous to automatically overwrite preferences. Maybe the best solution is to fail in `main.c` if `nvs_flash_init` fails as we don't expect to operate without a functional NVS. The developer can decide the appropriate resolution, as there isn't one that seems always correct.

```
ESP_ERROR_CHECK(nvs_flash_init());
```

zBuffer, 2019-05-15:
Agreed. An error thrown will be good enough to alert the developer.

closed 2019-05-15 by zBuffer