Moddable-OpenSource/moddable #193: Failed to allocate bytes for chunk error

praveenravichandran, 2019-05-16:
Am getting the following error when I tried to burn the application in esp32 board,
!!! xs: failed to allocate 64 bytes for chunk !!!
C:\Users\Praveen\Projects\moddable\xs\platforms\esp\xsHost.c (762) # Break: C: xsDebugger!
# Chunk allocation: failed for 64 bytes
unhandled exception

Team, could you please help on how to overcome this issue.

phoddie, 2019-05-16:
It looks like your script has run out of memory.

Please confirm that your build is working properly by verifying that the [helloworld](https://github.com/Moddable-OpenSource/moddable/blob/public/examples/helloworld/main.js) example works on the device.

The Instruments panel of our xsbug JavaScript debugger is useful for monitoring the memory use of your scripts. It shows real time use of the JavaScript slot heap, the JavaScript chunk heap, the JavaScript stack depth, and system memory.

praveenravichandran, 2019-05-16:
Thanks much for the quick response phoddie. Helloworld example is working perfectly. Is there any option to increase the chunk, could you please help us out on how to increase the chunk memory.

phoddie, 2019-05-16:
> Helloworld example is working perfectly

Perfect.

You can adjust the allocations of the JavaScript virtual machine using the `creation` object in the manifest. It looks like you made an attempt to do that above by adjusting the `chunk` sub-object. The value you adjusted is the initial size of the chunk heap within the `static` allocation. Both the slot and chunk heaps live within the size indicated by `static`. The static allocation defaults to 32 KB in the [manifest_base.json](https://github.com/Moddable-OpenSource/moddable/blob/public/examples/manifest_base.json#L8) included by all projects.

	"creation": {
		"static": 32768,
		"chunk": {
			"initial": 1536,
			"incremental": 512,
		},
		"heap": {
			"initial": 512,
			"incremental": 64,
		},
		"stack": 256,

To double the memory available to the virtual machine, set `static` to 65536. You do not usually need to adjust the initial values of `chunk` or `heap`. Those heaps will grow as needed by your script.

On ESP32 today, the largest value your can use for static is about 128 KB.

FYI - The chunk and heap initial values and increments are present ot allow for optimization when you do know the baseline memory use of your application. It is

praveenravichandran, 2019-05-16:
So that means if I changed all those parameters in manifest base file should do the trick?

phoddie, 2019-05-16:
Yes and no.

You should only change `static`. But you should not edit `manifest_base.json`. Ever. You should override it in your project's manifest.

"creation": {
	"static": 65536,
},

That said... increasing allocations to overcome out of memory problems is not a sustainable strategy. Eventually, you won't have any more memory left to claim. You should work on minimizing the memory use of your code. That's why I referred  you t the Instruments panel in xsbug above. The largest static allocation we have used in a real product is 42 KB. And that is for a very complex product. ;) There are, of course, valid reasons to need more, but in embedded development is a best practice to be conscious of memory use during all stages of development.

praveenravichandran, 2019-05-16:
Agreed, actually we have tried to minimize the size of an application then finally we made it to 48 kb. At that time we got the same chunk allocation error when we are initiali

praveenravichandran, 2019-05-17:
Thanks, it worked right now will try more on reducing the size.

closed 2019-05-17 by praveenravichandran