Moddable-OpenSource/moddable #196: define: undefined variable

Ethirajulu, 2019-05-17:
Team, we are getting the "define: undefined variable" error when running the following code.

`(function (global, factory) {
  if (typeof define === "function" && define.amd) {
    define(["module", "exports"], factory);
  } else if (typeof exports !== "undefined") {
    factory(module, exports);
  } else {
    var mod = {
      exports: {}
    };
    factory(mod, mod.exports);
    global.scion = global.scion || {}; global.scion.core = mod.exports;
  }
})`

This is a function from a node package that we are trying to integrate into Moddable. It works fine in Node environment.

Could you please suggest us a solution.

Is there is a way to convert the npm packages into XS engine understandable format?

phoddie, 2019-05-17:
I pasted the code above into example/helloworld.js. It compiles and runs without error.

The Moddable SDK is not an emulation of the Node runtime. The Moddable SDK is optimized for embedded devices with orders of magnitude fewer resources than the devices Node is optimized for. They are necessarily different. While they share the same core JavaScript language, they are not the same. For example, all the code in the Moddable SDK runs from true ECMAScript modules which means, among other things, that all code runs in JavaScript strict mode.

You generally cannot simply drop in a Node module and expect it to work as-is. At a minimum some adjustments are likely needed. Sometimes it just won't work because it has too many dependencies on the Node runtime.

@lprader wrote a [good blog post](http://blog.moddable.com/blog/replacing-web-libraries/) about the topic of reusing existing libraries that you might check out.

closed 2019-05-17 by phoddie

Ethirajulu, 2019-05-20:
Thanks phoddie. Below is the link to the JS file that we trying to integrate with moddable. 

[scion.txt](https://github.com/Moddable-OpenSource/moddable/files/3197884/scion.txt)

We are facing the "define: undefined variable" error when loading this file. Could you please help us out here with a solution.

phoddie, 2019-05-20:
As noted above, we do not recommend importing Node.js modules for use in embedded JavaScript. If you choose to attempt to do so, you should expect challenges that require expertise in JavaScript and embedded development to resolve.

In this particular case, the problem is that the code expects `this` to be `global` when the script launches. That is the case in sloppy mode. In strict mode, `this` is `undefined`. As noted, the Moddable SDK  always runs in the more modern strict mode. Change line 13 of the code from

```
})(this, function (module, exports) {
```

to this

```
})(global, function (module, exports) {
```

That allows the code to run without an exception and `scion` to be added to `global`.

This code is huge for an embedded library (nearly 4000 lines!) and maybe in imports more (`require('./support/isBuffer')` and `require('inherits');` and `require('scion-core-base')`).  Those 4000 lines include a lot of boilerplate added by a packager. It really doesn't make sense to have code like `old school shim for old browsers` in embedded libraries.

Just launching the code consumes around 75 KB of RAM.  [Preloading](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/preload.md) will reduce that though it seems likely the library will require more to do useful work. I'll be impressed if this works well, if at all, on an embedded device.  But perhaps if add an external RAM chip to an ESP32 and are a bit luck, it will.

Ethirajulu, 2019-05-21:
Thanks a lot phoddie. We will try this out and let you know the result :)