Moddable-OpenSource/moddable #197: moddable-one with Arduino IDE

jpasqua, 2019-05-19:
I have a couple of moddable-one units for use with the moddable sdk, but I realized that this hardware would be a good upgrade for an existing esp8266 project I already developed using the arduino IDE. Rather than port my C++ code over, I'd like to be able to use a moldable-one directly with the Arduino IDE and my existing code and libraries. Is there a way to do that? Will it retain use of the touchscreen?

phoddie, 2019-05-20:
You can run code form the Arduino IDE on a Moddable Zero. We do that from time to time. If you have Arduino compatible libraries for the display and touch drivers, those should work.

That said, Moddable only provides support for using the Moddable Zero with the Moddable SDK.

jpasqua, 2019-05-20:
Thanks Peter. According to the [moldable-one docs](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/moddable-one.md), it has a "240 x 320 QVGA IPS display driven by a MIPI Display Serial Interface compatible display controller with an FT6206 capacitive touch controller". I assume that it is an ILI9341 and that the wiring is the same or similar to what is shown for the moddable-zero. If so, I do have libraries that will work.

BTW, I just got the moddable-one units this past weekend, so I am new to them.

phoddie, 2019-05-20:
You've got the components about right. ILI9341 is a MIPI compatible display controller, yes. So, a driver for that works. The wiring details/configuration you need are in the manifest for the [Moddable One build target](https://github.com/Moddable-OpenSource/moddable/blob/public/build/devices/esp/targets/moddable_one/manifest.json) and the manifests it includes (e.g. ILI9341 and FT6202).

closed 2019-07-11 by mkellner