Moddable-OpenSource/moddable #198: Moddable SDK - Windows Host Environment Setup - build error

nystrom68, 2019-05-20:
I'm having a problem setting up the Moddable Host Environment in Windows. I'm following the instructions from [here](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md)

When I launch the build from the Developers Command Prompt, the error I'm getting is:
`C:\Users\<usr>\Projects\moddable\build\makefiles\win>nmake GOAL=debug BUILD_DIR=C:\Users\<usr>\Projects\moddable\build XS_DIR=C:\Users\<usr>\Projects\moddable\xs /c /f tools.mak /s 
LINK : fatal error LNK1181: cannot open input file 'C:\Users\<usr>\Projects\moddable\build\tmp\win\debug\lib\xsa.c.o'`

I have substituted <usr> for my windows username above and in the attached file.

I have tried cloning using git and have downloaded the directory, with similar results.

The full output is attached.
[build_output.txt](https://github.com/Moddable-OpenSource/moddable/files/3196358/build_output.txt)

wilberforce, 2019-05-20:
<usr> In the path  should be your user name -  is it Michael?

You can check the environment vars with `env` command. If you set after the command prompt is open the current session might not b e set - quit the cmd shell and start another.

bfriedkin, 2019-05-20:
@nystrom68 -

We apologize for the inconvenience.

Unfortunately there was an error in the Windows tools make file that led to the problem you bumped into. We pushed a fix. Please update your Moddable SDK and try building again.

Regards,
Brian

nystrom68, 2019-05-21:
Thanks, works for me now.

Regards, Mike

closed 2019-05-21 by nystrom68