Moddable-OpenSource/moddable #200: Resource creation for png files

wilberforce, 2019-05-23:
I'm using resources as file source for a web server. I have found the with png files they are always split into the alpha channel and rgb, whereas I would like to have switch to override this behaviour.

Could it be added as flag in the resources section of the manifest to pass to:

https://github.com/Moddable-OpenSource/moddable/blob/public/tools/mcrez.js ?

phoddie, 2019-05-24:
Yes, this is imperfect in the tools. The tools perform transformations on resource to adapt them for use on the target microcontroller. This is great for rendering graphics, but not what you want when serving the data from web server. An easy workaround is to change the extension of your PNG files to from `.png` to `.bin`.

wilberforce, 2019-05-30:
@phoddie

Will this work for pngs?

https://github.com/Moddable-OpenSource/moddable/commit/ee868d5c4904577d08e505d14de4362b1c1322c7

What is the data section? I don't think this is documented.

phoddie, 2019-05-30:
Yes, it will work for `png` files and all other  files extensions. The `data` section is new. It is a way to include untransformed files in a resource. The documentation will be updated.

wilberforce, 2019-05-30:
Fixed https://github.com/Moddable-OpenSource/moddable/commit/ee868d5c4904577d08e505d14de4362b1c1322c7

closed 2019-05-30 by wilberforce