Moddable-OpenSource/moddable #201: File: file not found! unhandled exception

praveenravichandran, 2019-05-23:
Team,
I have implemented the file concept in my project. Implementation is working fine in windows simulator but It is not working in the esp32 target. Am getting the below error in modFIle. C 
\modFile.c (76) # Break: File: file not found!. I have added the file configuration in manifest as below `"esp32/moddable_zero": {
      "modules": {
        "*": "$(MODDABLE)/modules/files/file/esp32/*"
      },
      "config": {
        "file": {
          "root": "./"
        }
      }
    }`

Could you please help on this

wilberforce, 2019-05-23:
The root would need to be "/spiffs/". If you have a recent build I think this is now the default.

praveenravichandran, 2019-05-23:
So you mean that I should give "/spiffs/" instead of ./ in root.? Please correct me if I'm wrong

lprader, 2019-05-23:
> So you mean that I should give "/spiffs/" instead of ./ in root.?

That's correct.

The [files manifest](https://github.com/Moddable-OpenSource/moddable/blob/public/modules/files/file/manifest.json) includes the root for each target platform in the `config` object. You can access this in your script as done in the [files example](https://github.com/Moddable-OpenSource/moddable/blob/public/examples/files/files/main.js#L18).

praveenravichandran, 2019-05-24:
Thanks much Iprader and Wilberforce. Will try that and update here

praveenravichandran, 2019-05-24:
We tried adding the "/spiffs/" in the manifest and also declared the root config. Still the error is persists

lprader, 2019-05-24:
Try running the `files` example. It works for me on the ESP32. Does it work for you?

praveenravichandran, 2019-05-24:
Will check that

lprader, 2019-05-24:
If the `files` example doesn't work for you, let me know. Otherwise, follow the same pattern it uses to get the correct root by taking these steps:

1. In your app's manifest, include the `files` manifest.

	```	
	"include": [
		/*other includes here*/
		"$(MODDABLE)/modules/files/file/manifest.json",
	],
	```

2. In your app's script, import the `config` object.

	```
	import config from "mc/config";
	```

3.  Use `config.file.root` as the root. Make sure you include the root in file names.

	```
	const root = config.file.root;

	let file = new File(root + "example.txt");
	```

praveenravichandran, 2019-05-24:
I think file example is working. The word think is put here because it doesn't show any error while burning the application and nothing was showing in the xsbug logs too 😁.

lprader, 2019-05-24:
Hmm...you should see messages traced to the `xsbug` console when you run the example. Make sure you're running a debug build. To do that, add `-d` to your build command. For example:

```
mcconfig -d -m -p esp32
```

praveenravichandran, 2019-05-24:
Yeah, I always used to run in debug mode.

lprader, 2019-05-24:
Based on your original comment, it sounds like you see messages (or at least errors) traced to the console when running other applications. Is that true? 

And do you hit breakpoints in other examples? For example, if you run `helloworld`, do you hit the breakpoint and then see "Hello, world - sample" traced to the console after you press the run button?

praveenravichandran, 2019-05-24:
Exactly!

lprader, 2019-05-24:
Can you share a screenshot of `xsbug` when you run the `files` example on the ESP32 using these commands?:

``` 
cd $MODDABLE/examples/files/files
mcconfig -d -m -p esp32
```

You should get a machine tab (like the one circled in red in the image below). When you click that tab, you should see the instrumentation panel (and it should be updating the values of each row every second or so).

[image]

praveenravichandran, 2019-05-25:
Yeah I will post it here.

praveenravichandran, 2019-05-27:
[image]

lprader, 2019-05-28:
Sorry, now I'm a little confused--you originally said:

> it doesn't show any error while burning the application and nothing was showing in the xsbug logs too

That screenshot clearly shows messages traced to the console and an error. So that doesn't seem to be a problem. Is that correct?

praveenravichandran, 2019-05-28:
Iprader you are right, Initially I didn't found any of the trace logs as mentioned above. But after you asked for the screenshot I just try by changing the environment variable path of xsbug that made the debugger .There were something weird at that time. I suspect that I didn't get any error message because xsbug is not launched at that time. However my main issue in the board file not found exists.

lprader, 2019-05-28:
Alright, gotcha. Glad that part is solved.

As for the "file not found" issue, what kind of ESP32 board are you using? In your very first comment you mentioned you added a `config` object for the `esp32/moddable_zero` target in your manifest. That target is meant for use with an [ESP32 paired with a SPI display](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/displays/wiring-guide-generic-2.4-spi-esp32.md). Is that what you're using?

praveenravichandran, 2019-05-29:
Exactly

lprader, 2019-05-29:
Would you mind sharing your application so I can see if it works for me?

praveenravichandran, 2019-05-29:
[files.zip](https://github.com/Moddable-OpenSource/moddable/files/3230834/files.zip)

wilberforce, 2019-05-29:
@praveenravichandran 

When did you get your Moddable sources? - they could be out of date and changes made since you pulled your copy of the repository...

praveenravichandran, 2019-05-29:
Might be, I have the code which has been last updated on May 1st.

praveenravichandran, 2019-05-29:
Meanwhile, I don't see any commits regarding the file issues @wilberforce

lprader, 2019-05-29:
Ah-ha. I understand now.

You are attempting to read from a file that doesn't exist on the device. You have a `preferences.json` file in your source folder, but you don't specify in your manifest that it should go on the device.

lprader, 2019-05-29:
I'm not sure how to include a JSON file in your build (I'll look into that tomorrow), but if you want to store preferences like the ones in `preferences.json`, I recommend you use the [Preference class](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/files/files.md#preference) anyways.

praveenravichandran, 2019-05-29:
Okay, So that means If I read and write in a single program it doesn't requires any configuration in manifest and it automatically takes from the root path. Correct me if my understanding wrong

praveenravichandran, 2019-05-29:
Thank you Iprader, Will definitely go through that preference class too.

lprader, 2019-05-29:
> Okay, So that means If I read and write in a single program it doesn't requires any configuration in manifest and it automatically takes from the root path. Correct me if my understanding wrong

No files are ever added to the device automatically. You have to specify which files go onto the device in the manifest. Or you can create them at runtime, as the `files` example does.

From the [File class constructor documentation](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/files/files.md#constructorpath--write):
If the file does not exist, an exception is thrown when opening in read mode. When opening in write mode, a new file is created if it does not already exist.

If you look at the source for the `files` example, you can see there are no files other than `main.js` and `manifest.json`. But in `main.js`,  an instance of the `File` class is created with the `write` argument set to `true`. Because this file does not exist, it is created. Then it is written to.

```
file = new File(root + "preferences.json", true);
file.write(JSON.stringify(preferences));
file.close();
```

praveenravichandran, 2019-05-29:
Got it!

praveenravichandran, 2019-05-29:
@lprader  You were right. It is working when I tried it by deleting the get data from preference part alone.

praveenravichandran, 2019-05-29:
For taking the external JSON file, whether we can try it out just by giving the folder name where the JSON file is put under modules of manifest .

phoddie, 2019-07-11:
Closing, as this appears resolved.

closed 2019-07-11 by phoddie