Moddable-OpenSource/moddable #203: # Break: RegExp.prototype.flags: dead strip!  error, during esp32 build

Priyalakshmi-Laxmanan, 2019-05-27:
Hi Team,

In reference to this issue https://github.com/Moddable-OpenSource/moddable/issues/196, I am trying to run the moddable after integrating with the third party scion.js file.

This is running perfectly in simulator mode (mcconfig -d -m -p win), but while tryinging to do a build for ESP32 setup, we are getting this error # Break: RegExp.prototype.flags: dead strip! 

In the documentation, we found the strip option in manifest file, is there any way to simply just exclude the functions for striping in the array instead of including all other needed functions.

phoddie, 2019-05-27:
If `RegExp` is working in general (and I presume it is, as you did not report otherwise), but `flags` is failing, try referencing that explicitly in your project's main.js:

```
let foo = RegExp.prototype.flags;
```

If that does not resolve the problem, additional details will be required.

Note that in  `manifest_base.json` there is no different in how stripping is applied by platform. Is it possible you have overridden this with a "strip" option in your manifest?

Priyalakshmi-Laxmanan, 2019-05-28:
Hi Phoddie,

Thanks!
In our code we didn't used `flags` explicitly, the following in the code which caused 'dead strip' error,

```
var RX_WHITESPACE = /\s+/;
t.events = t.event.trim().split(RX_WHITESPACE);
```
Currently we have fixed this with a workaround,

`t.events = t.event.trim().replace(/  +/g, ' ').split(' ');`

No changes in the `manifest_base.json` is done, for the striping.

Also require one more information, 
Could you please tell us, while flashing, from which path/binary file will be used for flashing, to give an insight on the build file sizes.

phoddie, 2019-05-28:
Thank you for the details. Now the problem is clear. We'll look at a solution. In the meanwhile, it looks like you have a workaround.

> Could you please tell us more information about flashing, during flashing which path/binary file will be used for flashing into the device, to get more insight about build file sizes

The flashing process uses the unmodified ESP IDF. The documentation from Espressif explains the details. The build tools report the binary image size when flashing. The binary is located at `$MODDABLE/build/bin/esp32/YOUR_BUILD_TARGET/debug/xs_esp32.bin`.

N.B. This is a distinct topic from the original issue. If you wish to discuss it further, please open a separate issue.

Priyalakshmi-Laxmanan, 2019-05-29:
Thanks for the response Phoddie!

We will wait for the your response regarding this issue.

For other topics, I will open a separate issue

phoddie, 2019-08-05:
Fix committed.

closed 2019-08-05 by phoddie