Moddable-OpenSource/moddable #205: Getting Started with the Simulator

CaseyB, 2019-06-03:
I went through the Getting Started section but they all assume you have the board plugged in via USBso they fail with the following error:

[Errno 2] could not open port /dev/cu.SLAB_USBtoUART: [Errno 2] No such file or directory: '/dev/cu.SLAB_USBtoUART'

I couldn't find anything that explained how to use the simulator. When I run the ScreenTest app I also can't figure out how to configure it to simulate the Moddable Three. I would be grateful if you could point me in the right direction.

lprader, 2019-06-03:
The main [Getting Started guide](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md) has instructions for running on the simulator to test that setting up your host environment worked. Basically you can just leave out the platform argument and `mcconfig` defaults to running for the simulator, i.e.

```
cd $MODDABLE/examples/helloworld
mcconfig -d -m
```

Documentation of the simulator is tucked in the [tools documentation](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/tools/tools.md#simulator). To simulate Moddable Three, change the size of the window to 250x122.

[image]

CaseyB, 2019-06-03:
Great! Thank you for the update! It tried a couple of the screen sizes but it just changed the size but still showed the same device. It might be nice to re-label them to show the device name too.

closed 2019-06-04 by CaseyB