Moddable-OpenSource/moddable #207: Multiple pwm objects

diogoviannaaraujo, 2019-06-10:
Hi,

pins/pwm module expose a object but if you create multiple instances of it they all share the same channel.

Maybe it could receive a optional channel option in constructor together with 'pin'.
Or maybe it could keep track of available channels in an varaible and get the next available.

phoddie, 2019-06-10:
Yes. We ran into this a couple weeks ago too. Apologies. Your analysis of the problem is exactly correct. We'll get a fix in place.

diogoviannaaraujo, 2019-06-10:
I have been able to maek it work based on pulsecount implementation:

https://github.com/diogoviannaaraujo/moddable/blob/esp32-pwm-channels/modules/pins/pwm/esp32/pwm.c

Just need to find the correct way to correctly destruct it

diogoviannaaraujo, 2019-06-10:
I guess i got destructing to work.

I still have some doubts about why the use of the initialized variable and if we should use Port as the channel.

phoddie, 2019-08-05:
There is a fix for this committed now. It supports up to four channels by default. That can be adjusted up or down using the project manifest.

closed 2019-08-19 by phoddie