Moddable-OpenSource/moddable #210: what sort of free file space should I expect on a 4M esp32 with an Xs7 build installed

jim80, 2019-06-12:
My esp32 is reporting free file space of only 52961 bytes!
{"fileSystemInfo":{"total":52961,"used":45180,"maxPathLength":31},"fileListing":[{"filename":"test.jpg","size":44516}]}
This doesn't seem right to me! Test.jpg was a whole lot bigger when uploaded (that caused a write failure error, from moddable file module, which seems to confirm the reported size is accurate...)

I'll just ask the question, does that seem right to you?

esptool.py reports a 4M flash, by the way.

I appreciate this may not even come under the realm of the Moddable project and it could be a separate hardware.config problem, but any clue as to where to look to solve it would be much appreciated. I'm not quite sure where to start just now...

heres the output from mmconfig, by the way, uploaded file sizes look reasonable to me....

# xsc main.xsb
# xsl modules
# cc mc.xs.c (slots in flash)
/tmp/ccic895U.s: Assembler messages:
/tmp/ccic895U.s:41344: Warning: setting incorrect section attributes for .rodata.mod.1
# ld xs_esp.bin
CC /home/parallels/Projects/moddable/build/tmp/esp32/debug/idf/main/main.o
AR /home/parallels/Projects/moddable/build/tmp/esp32/debug/idf/main/libmain.a
LD /home/parallels/Projects/moddable/build/tmp/esp32/debug/idf/xs_esp32.elf
esptool.py v2.6
Flashing binaries to serial port /dev/ttyUSB0 (app at offset 0x10000 )...
esptool.py v2.6
Serial port /dev/ttyUSB0
Connecting........_
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: cc:50:e3:af:ba:0c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 13616 bytes to 8910...
Wrote 13616 bytes (8910 compressed) at 0x00001000 in 0.1 seconds (effective 1043.3 kbit/s)...
Hash of data verified.
Compressed 846080 bytes to 522145...
Wrote 846080 bytes (522145 compressed) at 0x00010000 in 7.3 seconds (effective 931.5 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 120...
Wrote 3072 bytes (120 compressed) at 0x00008000 in 0.0 seconds (effective 3850.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

wilberforce, 2019-06-13:
The default partition used in the build has a really small spiffs partition.

you need to update your `partitions.csv`: put this in the root of your project:

```
#nvs,      data, nvs,     0x9000,  0x006000,
#phy_init, data, phy,     0xf000,  0x001000,
#factory,  app,  factory, 0x10000, 0x3B0000,
#xs,       0x40, 1,       0x3D0000, 0x010000,
#settings, data, 1,       0x3E0000, 0x010000,
#storage,  data, spiffs,  ,        0x0F000,

nvs,		data,	nvs,	0x9000,		0x006000,
phy_init,	data,	phy,	0xf000,		0x001000,
factory,	app,	factory,0x10000,	2048K,
storage,	data,	spiffs,	0x300000,	512K,
xs,			0x40,	1,		0x3D0000,	0x010000,
settings,	data,	1,		0x3E0000,	0x010000
```
The commented out settings are the defaults - the example increased to 512K.

Then in your manifest:

```
"esp32": {
			"build": {
				"PARTITIONS_FILE": "./partitions.csv",
			},
```

this overwrites the default set up with the local partitions file.

jim80, 2019-06-13:
Oh superb, so many thanks @wilberforce , really appreciated!
I'll not get to it right now, it being late here, but I'll take a look tomorrow.
Thanks Again!

jim80, 2019-06-13:
That'll give me a 512K spiffs files system I guess? It's enough for me I should think!

jim80, 2019-06-13:
Info on partitions : https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/partition-tables.html

jim80, 2019-06-13:
Not working for me :( I'm missing something...

wilberforce, 2019-06-13:
You need to make sure that partition.bin gets rebuilt - delete this file under build and see if it gets re-made - otherwise you need to delete your esp32 tmp build folder to force it to get rebuilt

wilberforce, 2019-06-13:
How do you do a xs7 build?

jim80, 2019-06-13:
By Xs7 I mean moddable!

jim80, 2019-06-13:
I did try deleting the esp32 tmp build folder, saw the project being rebuilt, but no joy.
Also ending up with this in the root of my manifest. json, after having a look at some of the others.
"platforms": {
		"esp32": {
			"build": {
				"PARTITIONS_FILE": "./partitions.csv"
			}
		}
	}

It could be that's incorrect though, I'd tried just 

		"esp32": {
			"build": {
				"PARTITIONS_FILE": "./partitions.csv"
			}
		}
But I don't recall if I'd deleted the temp build directory when I did that.

It's not the biggest of issues for me right now, but would be nice to get it working!
I'm using linux, but I'd expect it to work in the same way...

jim80, 2019-06-13:
So, neither option seems to have any effect, even after deleting the tmp/esp32 directory for a full rebuild.
I did find partitions.csv in moddable/build/devices/esp32/xsProj, tried your example there.
That did have an effect, of esp32 then crashing when I tried to read the filesystem! Dunno, I guess I need to know before before messing with the flash!
No problem, got it back to how it was, but would still like to have some control over this.

wilberforce, 2019-06-14:
You probably need to erase the flash of the esp32 to ensure that the flash is reset.

jim80, 2019-06-14:
So I need to learn how to do that! Thanks @wilberforce , much appreciated. Apologies for not being all that informed on the esp platform, but your help is most welcome. I'll get myself another dev board before I (maybe) brick the only one I have, I'm just a little afraid to right now since I'd rather be writing javascript than googling how to restore an esp flash ;)

phoddie, 2019-06-14:
FWIW - you are unlikely to brick your device by modifying the partitions. You can reset everything with `esptool.py erase_flash`.

The information @wilberforce provided appears correct. It isn't obvious why isn't working for you, though I haven't tried it myself. Our default builds all use the `partitions.csv` you found in $MODDABLE/build/devices/esp32/xsProj/partitions.csv`. To experiment, there's no harm in modifying that. But, for real projects the approach of creating your own and referencing it from your project manifest, as @wilberforce described, is preferred.

You might try making a simple modification to `partitions.csv` and then run helloworld just to see that all the fundamentals are well before trying to use the file system.

wilberforce, 2019-06-14:
I seem to recall going through the same drama. In the end I think might have used the esptool to make a new partition.bin and repacae that as part of the build.

This might have been part of the reason to moving to using resources for the web assets rather than using a file system in my case... and there have been resent changes here with the idea of a `data` resource so things like png don't get modified.

bfriedkin, 2019-07-08:
Hello -

Closing the loop here. I've attached a basic `filespace` example app that incorporates the `partitions.csv` file @wilberforce provided. When run the example outputs the following to the `xsbug` console:

```
Used/Total: 0/474641
```

Before running the app, I erased the ESP32 flash using the following command:

```
python $IDF_PATH/components/esptool_py/esptool/esptool.py erase_flash
```

Regards,
Brian

[filespace.zip](https://github.com/Moddable-OpenSource/moddable/files/3369217/filespace.zip)

closed 2019-07-08 by bfriedkin