Moddable-OpenSource/moddable #212: Debugging BLE

diogoviannaaraujo, 2019-06-13:
Hi,

Im having a problem where iOS bonding is not working, the connection works but device is never authenticated and keys dont persist.

In espressif forums I see people talking about some idf logs but i cannot access them.

Is there some way to receive idf log messages on xsbug or in terminal?

bfriedkin, 2019-06-13:
Diogo -

Logging ESP-IDF Bluetooth messages from Moddable apps is possible, but can be tricky, and often times don't provide enough information to pinpoint the problem at hand. Let's first try to investigate without that.

The first thing I'd try is removing the Moddable peripheral device from the iOS Bluetooth cache. Sometimes pairing fails on iOS when you change the peripheral app, because the iOS cache contains stale information. On the iOS device, in Settings -> Bluetooth, toggle Bluetooth off and on. If your device is still listed, tap the "i" button and then tap "Forget This Device". Finally try pairing again with the ESP32 Moddable BLE app.

Another problem I have observed when trying to establish a bonded relationship between an iPhone and ESP32 Moddable BLE app, is that iOS seems to be picky about what types of devices it is willing to bond with, especially when you pair from the iOS Settings app. Perhaps the Core Bluetooth documentation and/or iOS developer forums can provide additional details?

To try and pinpoint this problem, I've attached a simple Moddable HID peripheral app that does pair and bond with my iPhone. I pair from the iOS Settings app and the Moddable device is "remembered" in the device list. When I unplug the ESP32, the device shows as "Not Connected." When I plug the ESP32 back in, the device displays as "Connected" and no pairing prompt is displayed. Hence long term keys have been exchanged and stored on both devices. It seems that deploying the HID service from the Moddable app is required in order to pair/bond from the Settings app. Using this working app as a starting point, you might be able to more easily resolve the issue in your app.

Lastly, there is BLE logging in the Moddable BLE server and client implementations. In your [modBLEServer.c](https://github.com/Moddable-OpenSource/moddable/blob/public/modules/network/ble/esp32/modBLEServer.c) file, you can enable the `LOG_GATTS` and `LOG_GAP` defines:

```
#define LOG_GATTS 1
#define LOG_GAP 1
```

Native ESP32 BLE callback events will be traced to the xsbug log. You can compare the event sequence between the working attached app and your app looking for clues.

Regards,
Brian

[bonding.zip](https://github.com/Moddable-OpenSource/moddable/files/3287662/bonding.zip)

diogoviannaaraujo, 2019-06-13:
Hi, i have some aditional details about this problem.
Its not all iOS devices, I have an iPhone 7 Plus and an iPad Pro, both cannot bond with the esp32 but a friends iphone 7 is bonding...

Im not trying to bond from settings, since the BLE devices does not show there, it asks for bonding after connection, im using Nordic's app for BLE testing.

I've tryed to erase all network settings on iOS and then try again but no go.. Since the only shared variable between the failures are that they all are my devices I wonder if apple share Bluetooth LTK between all my devices, and that is causing some bonding problems.

Will do some more tests with your HID example and go throu the logs and will get back to you tomorrow

bfriedkin, 2019-07-03:
Diogo -

Checking back in with you. Have you been able to narrow down the issue here?

Regards,
Brian

phoddie, 2022-10-20:
Closing due to inactivity. (Also, we have lots of projects successfully using BLE with iOS!)

closed 2022-10-20 by phoddie