Moddable-OpenSource/moddable #213: Software SPI for display driver on ESP32

meganetaaan, 2019-06-14:
Hi. I'm trying to use Moddable on [M5StickC](https://docs.m5stack.com/#/en/core/m5stickc), a new product from M5Stack.
I added new target files.
* [setup-target.js](https://github.com/meganetaaan/moddable/blob/public/build/devices/esp32/targets/m5stick_c/setup-target.js)
* [manifest.json](https://github.com/meganetaaan/moddable/blob/public/build/devices/esp32/targets/m5stick_c/manifest.json) including a driver init sequence ported from original library

It can run hello-world example. But when I run commodetto example the display still does not work.

After some debug I've found [the original library uses software spi through normal GPIO pins](https://github.com/m5stack/M5StickC/blob/568c35d62376031ee2de7b6742309af041eb8809/src/utility/In_eSPI.h#L355), while Moddable ili9341 driver(ESP32) uses [hardware spi](https://github.com/Moddable-OpenSource/moddable/blob/0aff5efc56213d7737b18e5c828fced3ab1b50ef/modules/pins/spi/esp32/modSPI.c#L112).
I guess this is the cause that my setting does not work.

My question here is
* Can Moddable use software spi to drive display or will it be able to as the future plan?
* If not, I'd like to fix it adding the option for sw spi. Do you have any suggestion about the design?

phoddie, 2019-06-15:
We don't plan to implement software SPI, at least not for hosts that have hardware SPI. It would be very unusual to find a piece of hardware that doesn't work with hardware SPI, but possible. We haven't seen one yet, but maybe this is the first. ;)  Still, I wouldn't assume it is impossible to make hardware SPI work. Sometimes it is necessary to change the configuration -- there are many options. @andycarle can you offer any suggestions here?

wilberforce, 2019-06-16:
I'm not sure that is using software spi - however with the esp32 I believe any pins can be mapped via the gpio matrix anyway...

Should this:

https://github.com/meganetaaan/moddable/blob/public/build/devices/esp32/targets/m5stick_c/manifest.json#L14

`"ili9341": {` be `"ili9341": {"st7735": {

Is this drive compatible?

https://github.com/Moddable-OpenSource/moddable/blob/26bb48a3857976b2be9a1087376914d4fa79c705/modules/drivers/ili9341/modIli9341.c

and here they also define the clock pin:
https://github.com/Moddable-OpenSource/moddable/blob/public/build/devices/esp32/targets/m5stack_fire/manifest.json#L19

And the spi clock is double what is on the  M5StickC page you linked.

meganetaaan, 2019-06-17:
Thank you for your quick feedback! Now I understand that software SPI is not reasonable and there should be the hardware way. I'll keep investigating.

I've fixed a property name of sck_pin and set frequency according to the spec.

https://github.com/meganetaaan/moddable/commit/ae474e0b91af6e7e533eb324036a5038e29bb288

But still the same (display does not work).

>"ili9341": { be `"ili9341": {"st7735": {

I read [st7735 document](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/drivers/st7735/st7735.md). I think this part("st7735" for screen and "ili9341" for definition) is right on my manifest. Is there misunderstanding?

wilberforce, 2019-06-17:
Where did you get your registers setup?

You don't need to add your own - as you are including this:
https://github.com/Moddable-OpenSource/moddable/blob/ed1cedf7021268d7fc229eedb05e62e6777c363e/modules/drivers/st7735/manifest.json

and this has the registered defined.

You only need to override the #defines that are different - the width/height and the pins. The hz is already 2000000 so you don't need to supply that.

So this should be all you need: 

```
{
	"include": [
		"$(MODDABLE)/modules/drivers/st7735/manifest.json",
	],
	"config": {
		"screen": "st7735",
		"touch": "",
	},
	"defines": {
		"spi": {
			"mosi_pin": 15,
			"sck_pin": 13
		},
		"ili9341": {
			"width": 80,
			"height": 160,
			"cs_pin": 5,
			"dc_pin": 23,
			"rst_pin": 18,
			"spi_port": "VSPI_HOST"
           },
	},
	"modules": {
		"*": [
			"$(MODULES)/pins/digital/monitor/*",
			"$(MODULES)/pins/digital/monitor/esp32/*",
		],
		"setup/target": "./setup-target",
	},
	"preload": [
		"monitor",
		"setup/target",
	],
}
```

meganetaaan, 2019-06-17:
I've found the cause of dark screen. Not the SPI problem but the backlight. Totally misunderstood. Actually the light has been off from the first time through my attempt...
M5StickC's display light is controlled from AXP192 power management unit. I did not initialize it in my Moddable code so display kept dark. I confirmed this behavior by first flashing M5StickC Android example, which includes codes for AXP192 turning on, then flashing Moddable codes keeping light on.

Sorry for bothering you all. And thank you so much helping me to make progress!
I close this issue. And after some more configuration I'm going to make pull request to add this new target.

closed 2019-06-17 by meganetaaan

wilberforce, 2019-06-17:
Perhaps rename the title at the the top of the page... For when others find it - it's not related to software Spi after all!

wilberforce, 2019-06-17:
Oh... And this was the first mention of a dark screen - that is a very big clue!

meganetaaan, 2019-06-18:
Renamed the title. sorry again. I got a lesson.

* Write a detail accurately. In my post "does not work" is too ambiguous.
* Do not haste a solution. find a root cause first.

wilberforce, 2019-06-18:
> Renamed the title. sorry again. I got a lesson.
Sorry - I was not trying to be harsh - I'm just an interested user of moddable and I'm looking at getting a M5 stick C or two now that you have pointed it out (thanks for that). I look forward to seeing your pull request when you have the port sorted.