Moddable-OpenSource/moddable #214: Question on esp8266 audioOut PDM pin

jim80, 2019-06-18:
From the docs 
"Defines for ESP8266
MODDEF_AUDIOOUT_I2S_PDM -- If zero, PCM samples are transmitted over I2S. If non-zero, samples are transmitted using PDM. Set to 32 for no oversampling, 64 for 2x oversampling, and 128 for 4x oversampling. Default is 0."

I take this to mean that setting a non zero value will result in a PDM output on a single pin.
Am I correct? If so, which pin will it be output to?

Many thanks!
Cheers!
JIm.

jim80, 2019-06-27:
https://www.hackster.io/janost/audio-hacking-on-the-esp8266-fa9464
tldr : the Rx pin, as far as I can tell, but I've not got around to testing it as yet.

phoddie, 2019-07-08:
Correct. When `MODDEF_AUDIOOUT_I2S_PDM` has a non-zero value audio is output in PDM format over a the I2S data pin. The I2S output pin is not configurable on the ESP8266. It is always GPIO 3. You can confirm the [pin use](https://github.com/Moddable-OpenSource/moddable/blob/public/build/devices/esp/lib/tinyi2s/tinyi2s.c#L170) in the I2S driver code.

closed 2019-07-08 by phoddie

jim80, 2019-07-09:
Thanks! I just had to look a little further than your docs, that's all.