Moddable-OpenSource/moddable #217: add troubleshooting section to "getting started" doc

boneskull, 2019-06-29:
I was having a tough time getting `helloworld` compiled, but found espressif/esp-idf#1330.  Turns out I had some conflicting stuff in my environment (one of `CPPFLAGS`, `LDFLAGS` and/or `LD_LIBRARY_PATH`).  

This should probably be noted somewhere in the docs.

jim80, 2019-07-01:
On a general note, moddable (I would say) is likely to attract people who like to code in js rather than c (like myself, with little c experience but some js) and maybe, dare I say it, less used to fixing any problems with the "getting started" procedure.
I had some problems myself which had their root in the 3rd party packages used and various dependencies but I'm usually reasonably good at googling error messages and working out the solution - which is what I did.
I do wonder how many people don't persevere with it, however?

boneskull, 2019-07-04:
@jim80 Agree.

The "getting started" is going to be overwhelming for a lot of people.  Most of those manual steps could likely be scripted (via a shell script or whatever), and the user could be prompted for input otherwise.

Another option might be a preconfigured VM to be run in VirtualBox or something.

bfriedkin, 2019-07-09:
@boneskull -

Thank you for pointing out this issue. We were able to reproduce the problem by setting `LDFLAGS` in the environment before building the Moddable app. Based on [espressif/esp-idf#1330](https://github.com/espressif/esp-idf/issues/1330), it seems that similar problems could occur when the environment defines other variables used by the Xtensa GCC toolchain.

Our `mcconfig` build effectively wraps the ESP32 build, by calling `make defconfig`, `make monitor`, `make flash`, etc... We were able to resolve this issue by unexporting `LDFLAGS`, `LD_LIBRARY_PATH` and `CPPFLAGS` from our build. Because the `unexport` directives are local to the Moddable build, the variables are maintained in the calling environment.

The fix will be available in our next Open Source update.

Regards,
Brian

closed 2019-07-10 by bfriedkin