Moddable-OpenSource/moddable #219: keyboard demo: `q` and `z` keys do not respond to touches correctly

zgramana, 2019-07-01:
When running the [piu/keyboard](https://github.com/Moddable-OpenSource/moddable/tree/public/examples/piu/keyboard) demo on a moddable-two, I'm unable to press either the `q` or the `z` keys.

- Attempts to press `q` appear to sometimes actuate the "back" arrow
- Attempts to press `z` often actuate the `a` key

wilberforce, 2019-07-01:
It sounds like you need to calibrate the touch screen

lprader, 2019-07-01:
@zgramana does it work any better if you run with a rotation of 180?

zgramana, 2019-07-02:
The `q` and `z` work, but the ⬆️ does not.

When I pressed the 🆗 button `xsbug` launched with break at `modules/base/timer/modTimer.c:136` with error message:
```
(host): invalid timer!
```
[image]

zgramana, 2019-07-02:
@lprader After playing with it a bit more at 180 rotated, I think shift is okay. I was likely just landing on the ⬆️ with the side of my finger pad.

I was able to duplicate the `invalid timer` at 90 rotation break again by hitting 🆗. After that, the dismiss-keyboard animation ran a little wonky.

lprader, 2019-07-03:
Strange that you're getting that `invalid timer` error and wonky animations. I can't seem to reproduce that, even by pressing various parts of the screen as quickly as I can. Do other apps that use touch work okay? Maybe try `piu/drag` or `piu/map-puzzle`.

Also, FWIW, we just added a new expanding keyboard with [horizontal](https://github.com/Moddable-OpenSource/moddable/tree/public/examples/piu/horizontal-expanding-keyboard) and [vertical](https://github.com/Moddable-OpenSource/moddable/tree/public/examples/piu/vertical-expanding-keyboard) variants, so you might try those too.

@phoddie any other ideas?

zgramana, 2019-07-04:
@lprader the `invalid timer` on dismiss issue has also happened when tapping at a leisurely rate. It appears non-deterministically AFAICT.

I tried rotating to 270 degrees as well, and in that case, the `p` key depresses the `l`, and depressing the `.` key presses 🆗.

The issue still occurs with the horizontal expanding keyboard as well.

No problems with `drag` or `map-puzzle`.

bfriedkin, 2019-07-05:
@zgramana -

Thank you for reporting the timer issue. We were eventually able to reproduce and fix the problem. The issue triggered when tapping backspace multiple times after characters were entered. The fix will be available in our next open source update.

Could you please clarify the issue you are having with the [horizontal-expanding-keyboard](https://github.com/Moddable-OpenSource/moddable/tree/public/examples/piu/horizontal-expanding-keyboard) app? That app does not use Timer and the manifest specifies the rotation angle. Hence there is no need to specify the rotation on the `mcconfig` command line. If there are issues with the horizontal-expanding-keyboard, please open a new case to cover those.

Regards,
Brian

zgramana, 2019-07-05:
@bfriedkin The timer issue I reported is ancillary to [the primary issue](https://github.com/Moddable-OpenSource/moddable/issues/219#issue-462461144) this ticket, the primary being that the `q` and `z` keys on both keyboards don't function correctly on my moddable-two. In other words, this issue _is_ that new case since it already covers keyboard problems.

I reported on the keyboard behavior with respect to rotation at [@lprader's request](https://github.com/Moddable-OpenSource/moddable/issues/219#issuecomment-507370619). While she only requested 180, I also included a report of behavior at 270 as well as it might prove helpful for diagnostic purposes.

phoddie, 2023-06-27:
Closing as this has been inactive for some time. Please reopen if there is new information. (FWIW – I tend to agree with @wilberforce that this looks like a calibration problem.)

closed 2023-06-27 by phoddie