Moddable-OpenSource/moddable #220: calling file.position on a closed file crashes the ESP :)

jim80, 2019-07-01:
Needless to say I did not do this deliberately!
However, it would have rather easier to debug my "not quite there" code if I'd got something other than a stack trace and reboot of my esp8266 ;)

Easily reproduced in the file example , thus ...
```
// writing/reading strings
file = new File(root + "test.txt", true);
file.write("This is a test.\n");
file.write("We can write ", "multiple", " values.\n");
file.write("This is the end of the test.\n");
file.position = 0;
let content = file.read(String);
trace(content);	
file.close();
trace("\n");

//CRASH AND BURN BABY!
trace (file.position)

```
which makes quite a nice accidental blink script as the built in LED flashes on restart :)

Anyway, just to let you know if this one is not already on your list, cheers!

jim80, 2019-07-01:
I've not tested any other silly calls to methods on a closed file, by the way. I guess if one were to fix .position you'd look at crash proofing the rest at the same time ...

bfriedkin, 2019-07-03:
@jim80 -

Indeed that is a creative way to reboot your device. :-)

Unfortunately, there isn't a way to "crash proof" calls to the `File` class once the file has been closed. `File.prototype.close()` calls the object destructor and the instance is effectively garbage collected. Hence your applications's `file` instance is no longer valid after calling `close()`.

The `File` class does throw exceptions when read or write operations fail on open files. The [files](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/files/files.md) documentation provides details.

Regards,
Brian

phoddie, 2019-07-12:
JavaScript calls aren't supposed to crash. I've added checks to the ESP8266 and ESP32 file modules to throw an exception if called after `close`. That will be included in our next push.

Thanks for the report.

ng-jim, 2019-07-14:
Many thanks Peter! That's more what I was thinking of, happy the report was some help.

commit e2e5223a (2019-07-17): file calls don't crash after close #220
closed 2019-07-17 by mkellner